ROBOTIS-GIT/turtlebot3 #1018: How to launch multiple ROS2 turtlebot3 in real world

zitongbai, 2024-05-04:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [x] etc (Please specify your ROS Version here)

ROS 2 Humble

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)
Jetson Orin nx

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [x] etc (Please specify your OS here)

Ubuntu 22.04

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [x] etc (Please specify your OS here)
Ubuntu 22.04

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

I want to launch multiple ROS2 turtblebot3 in the same network. However their topic names, tf names, etc. would conflict with each other. I found that namespace can be used for the ROS1 version, but I cannot find a easy way to do that in the ROS2 version. Is there any convenient method to achieve this? Thanks a lot.

krizchong, 2024-06-05:
You also implement this with namespace in ROS2!

starobi, 2024-12-03:
@GyuH13 the topics and nodes have namespaces but the frames don't. Do you know if that is problem? I was not able to change the name of all the frames and transformations

GyuH13, 2024-12-06:
@starobi 
Use the frame prefix.
[image]  

By using a prefix for each frame, you can differentiate between them and perform multi-robot operations without conflicts.  

[image]  

The /tf and /tf_static topics should not have namespaces.

If you have any additional questions, feel free to ask.

starobi, 2024-12-08:
Thank you for the response! I was able to plot the correct frames with the namespace. 

Now I want to execute AMCL with the namespaces. I created a launch file for that and adapted the param.yaml for the AMCL configuration but the initialization gets stuck and nothing happens if I set the initial pose. I noticed the logs registered when the AMCL node works correctly and the last step before working is LaserObjectCreated, but that doesn't happen when I initialize the AMCL node.  

I attach my frames, launch file and param.yaml file.

I would appreciate any feedback in the possible cause for my problem. Thanks! :)
[image]

[image]
[image]
[image]

GyuH13, 2024-12-19:
Hello. It seems like there is an issue with the synchronization of the stamp, but I’m not exactly sure what the problem is.
How about changing the 'scan_topic' parameter value to '/tb3_1/scan'?
This parameter determines the scan topic to subscribe to. If you leave the parameter as it is, the AMCL node won’t be able to subscribe to the scan topic, because the scan topic name is '/tb3_1/scan'.

GyuH13, 2025-02-20:
Since there are no further questions on this issue, we will close it.
For additional questions, please open a new issue.
Also, for namespace-related issues, please refer to the newly uploaded multi-robot example.

closed 2025-02-20 by GyuH13