ROBOTIS-GIT/turtlebot3 #1019: Turtlebot3 Unable to locate package ros-noetic-joy

3dokki, 2024-05-13:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [o] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [o ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [o ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [o ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [o ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - $ sudo apt-get install ros-noetic-joy ros-noetic-teleop-twist-joy \
  ros-noetic-teleop-twist-keyboard ros-noetic-laser-proc \
  ros-noetic-rgbd-launch ros-noetic-rosserial-arduino \
  ros-noetic-rosserial-python ros-noetic-rosserial-client \
  ros-noetic-rosserial-msgs ros-noetic-amcl ros-noetic-map-server \
  ros-noetic-move-base ros-noetic-urdf ros-noetic-xacro \
  ros-noetic-compressed-image-transport ros-noetic-rqt* ros-noetic-rviz \
  ros-noetic-gmapping ros-noetic-navigation ros-noetic-interactive-markers

8. Copy and Paste the error messages on terminal.

    - 
E: Unable to locate package ros-noetic-joy
E: Unable to locate package ros-noetic-teleop-twist-joy
E: Unable to locate package ros-noetic-teleop-twist-keyboard
E: Unable to locate package ros-noetic-laser-proc
E: Unable to locate package ros-noetic-rgbd-launch
E: Unable to locate package ros-noetic-rosserial-arduino
E: Unable to locate package ros-noetic-rosserial-python
E: Unable to locate package ros-noetic-rosserial-client
E: Unable to locate package ros-noetic-rosserial-msgs
E: Unable to locate package ros-noetic-amcl
E: Unable to locate package ros-noetic-map-server
E: Unable to locate package ros-noetic-move-base
E: Unable to locate package ros-noetic-urdf
E: Unable to locate package ros-noetic-xacro
E: Unable to locate package ros-noetic-compressed-image-transport
E: Unable to locate package ros-noetic-rqt*
E: Couldn't find any package by glob 'ros-noetic-rqt*'
E: Couldn't find any package by regex 'ros-noetic-rqt*'
E: Unable to locate package ros-noetic-rviz
E: Unable to locate package ros-noetic-gmapping
E: Unable to locate package ros-noetic-navigation
E: Unable to locate package ros-noetic-interactive-markers

9. Please describe the issue in detail.

    - PC settings are in progress according to the Robotis e-manual.
During the process, I get the above error in the "Installing Dependent ROS Packages" section. What should I do?

AnhHazz, 2024-05-14:
Maybe you haven't setup your computer to accept software from packages.ros.org yet. 
Try this in your terminal: 
`sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'`
`sudo apt install curl # if you haven't already installed curl`
`curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -`
then
`sudo apt update`

After those commands run your apt-get install again and see whether it solved

yun-goon, 2024-11-01:
Hi @3dokki 

It seems like an issue with the Noetic installation. Please try reinstalling it following the [manual](https://wiki.ros.org/noetic/Installation/Ubuntu) and then try again.
Thank you.

sunghowoo, 2024-12-06:
@3dokki This issue will be closed due to prolonged inactivity. If the issue persists, feel free to reopen it.

closed 2024-12-06 by sunghowoo