ROBOTIS-GIT/turtlebot3 #1022: Unable to create map using Gmapping

saad2121, 2024-06-06:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [x] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [x] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - $ roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping

8. Copy and Paste the error messages on terminal.

    - ERROR: cannot launch node of type [gmapping/slam_gmapping]: gmapping
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/mohammad.saad/catkin_ws/src
ROS path [2]=/opt/ros/noetic/share

9. Please describe the issue in detail.

    - I am trying to make the map using gmapping but unable to do it due to error mentioned above. I am following the tutorial from
    https://emanual.robotis.com/docs/en/platform/turtlebot3/slam_simulation/

farhan-haroon, 2024-08-22:
Please check if the following is fulfilled:

1. turtlebot 3 SLAM packages are installed and updated properly
2. workspace is sourced and built properly 
3. rosev=nvironment variables are sourced properly
4. check if the launch file has any errors

If all is good, please share a SS of the ~/.bashrc file and the gmapping launch file

yun-goon, 2024-11-01:
hi @saad2121 
Thank you for sharing the details of the issue. It seems that the gmapping package might not be installed or there may be an issue with the installation.

Please try reinstalling the gmapping package by running the following command:
```
sudo apt-get install ros-noetic-gmapping
```
After reinstalling, relaunch the SLAM process and see if the issue is resolved. Let us know if you continue to encounter problems.

sunghowoo, 2024-12-06:
@saad2121 This issue will be closed due to prolonged inactivity.  if the issue persists, feel free to reopen it.

closed 2024-12-06 by sunghowoo