ROBOTIS-GIT/turtlebot3 #1023: ros2 foxy error

Ashwath12052004, 2024-06-07:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    - HERE

Ashwath12052004, 2024-06-07:
need help with the first error
was tryna simulate dqn using turtlebot3 in gazebo11 using os foxy encountered this error specifically.

ros2 run turtlebot3_drl gazebo_goals
running on stage: 4, dynamic goals enabled: False
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
    return importlib.import_module(module_name, package=pkg_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'turtlebot3_msgs.turtlebot3_msgs_s__rosidl_typesupport_c'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ashwath/turtlebot3_drlnav/install/turtlebot3_drl/lib/turtlebot3_drl/gazebo_goals", line 11, in <module>
    load_entry_point('turtlebot3-drl', 'console_scripts', 'gazebo_goals')()
  File "/home/ashwath/turtlebot3_drlnav/build/turtlebot3_drl/turtlebot3_drl/drl_gazebo/drl_gazebo.py", line 263, in main
    drl_gazebo = DRLGazebo()
  File "/home/ashwath/turtlebot3_drlnav/build/turtlebot3_drl/turtlebot3_drl/drl_gazebo/drl_gazebo.py", line 74, in __init__
    self.task_succeed_server    = self.create_service(RingGoal, 'task_succeed', self.task_succeed_callback)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/node.py", line 1300, in create_service
    check_for_type_support(srv_type)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
    msg_type.__class__.__import_type_support__()
  File "/home/ashwath/turtlebot3_drlnav/install/turtlebot3_msgs/lib/python3.12/site-packages/turtlebot3_msgs/srv/_ring_goal.py", line 273, in __import_type_support__
    module = import_type_support('turtlebot3_msgs')
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 48, in import_type_support
    raise UnsupportedTypeSupport(pkg_name)
rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'turtlebot3_msgs'

yun-goon, 2024-11-01:
Hi @Ashwath12052004
Thank you for sharing the details of the issue. It seems that the turtlebot3_msgs package's type support (rosidl_typesupport_c) might be missing.

Please try installing the turtlebot3_msgs package by running the following command:
```
sudo apt install ros-foxy-turtlebot3-msgs
```
After reinstalling, try running your program again to see if the issue is resolved. 
Let us know if you continue to encounter problems.

closed 2024-12-06 by sunghowoo