ROBOTIS-GIT/turtlebot3 #1024: Running TurtleBot3 deactivates the sound output device and video playback of Youtube

MithraGhlm, 2024-06-20:
1. Which TurtleBot3 platform do you use?
    - Waffle

2. Which ROS is working with TurtleBot3?
    - ROS 2 Humble

7. Specify the commands or instructions to reproduce the issue.
    - $ ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py

8. Copy and Paste the error messages on the terminal.
    - I am not getting any errors from the Gazebo side. But certain things in my computer start malfunctioning.

9. Please describe the issue in detail.
   - I am using Gazebo 11.10.2 with Ros2 Humble with Ubuntu 22.04.4 on Docker.
When running Gazebo solely (e.i. nothing is spawned in gazebo), everything is fine and there's no problem. But when I run Gazebo with Turtlebot3 spawned in it, the sound output device of the computer doesn't stops working. Also all the Youtube videos on Chrome freeze  and I can not play them anymore. Though I can still brows websites.

Any help is highly appreciated.

yun-goon, 2024-11-01:
Hi @MithraGhlm 

Please note that there are many possible configurations in virtual machine environments, so we are unable to provide in-depth technical support.

However, running Gazebo in Docker could potentially consume a lot of CPU and memory, which might be causing the sound output issues.

We recommend reaching out to the ROS forum for more accurate assistance. Thank you for your understanding.

sunghowoo, 2024-12-06:
@MithraGhlm Please refer to @yun-goon  answer for initial guidance. This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly. Thank you for your understanding!

closed 2024-12-06 by sunghowoo