ROBOTIS-GIT/turtlebot3 #1026: Unable to Run SLAM on My PC

Sreedharbot, 2024-07-09:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ Y] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ Y] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [Y ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ Y] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [Y ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (ld08_driver/ld08_driver)

7. Specify the commands or instructions to reproduce the issue.

    - Not Able to run slam in my PC. 
    
 **Command  used:-**
 >  roslaunch turtlebot3_slam turtlebot3_slam.launch

8. Copy and Paste the error messages on terminal.

ros@ubuntu:~$ roslaunch turtlebot3_slam turtlebot3_slam.launch
... logging to /home/ros/.ros/log/c322866c-3de9-11ef-83c8-4336a2dc81b2/roslaunch-ubuntu-4818.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
RLException: Unable to contact my own server at [http://192.168.43.98:41011/].
This usually means that the network is not configured properly.

A common cause is that the machine cannot connect to itself.  Please check
for errors by running:

	ping 192.168.43.98

For more tips, please see

	http://wiki.ros.org/ROS/NetworkSetup

The traceback for the exception was written to the log file

10. Please describe the issue in detail.
Im using Raspberry pi as my master. and running roscore in my raspiberry pi. Able to run bringup and teleop on raspiberry pi via ssh. but not able to run slam on my pc. But i can view rosnode details from my PC. and able to execute rqt_graph from PC to see node communication. but still dont know Why I am not able to run slam.

yun-goon, 2024-11-05:
Hi @Sreedharbot 
It could indeed be an issue with the memory and CPU limitations of the Raspberry Pi. I recommend trying to set up the PC as the ROS master instead. 
Thank you.

sunghowoo, 2025-01-13:
@Sreedharbot This issue will be closed due to prolonged inactivity. If the issue persists, feel free to reopen it.

closed 2025-01-13 by sunghowoo

sue407, 2025-06-20:
Hi, we're in the same situation. able to run bringup and teleop on raspiberry pi but not able to run slam because of rviz. we already reinstalling packages. we are using dual boot of Windows and Ubuntu20.04. how can we solve it? 

[image]