ROBOTIS-GIT/turtlebot3 #1028: Turtlebot velocity is not updating

Faheemibrahim, 2024-08-07:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?
    - [ ] (ROS2 humble  hawksbill)

3. Which SBC(Single Board Computer) is working on TurtleBot3?
    - [ ] Raspberry Pi 4 B
    

4. Which OS you installed on SBC?
    - [ ] Ubuntu  Server 22.04.4 LTS (Jammy Jellyfish)

5. Which OS you installed on Remote PC?
 - [ ] Ubuntu  Desktop 22.04.4 LTS (Jammy Jellyfish)

6. Specify the software and firmware version(Can be found from Bringup messages)
    - Software Version: 2.1.0
    - Firmware Version: 1.2.3 

7. Specify the commands or instructions to reproduce the issue.
     ros2 launch turtlebot3_bringup robot.launch.py
     ros2 run turtlebot3_teleop teleop_keyboard 
     ros2 topic list /joint_states
     ros2 topic echo /odom

8. Copy and Paste the error messages on terminal.

[image]

9. Please describe the issue in detail.
I am not able to get the velocity from the turtelbot3 waffle_pi but when i use the ros2 topic echo/odom on the simulation im able to get the values for the velocity im pasting my yaml and launch file please do let me know if there are any corrections needed. 

**waffle_pi.yaml** 
turtlebot3_node:
  ros__parameters:

    opencr:
      id: 200
      baud_rate: 1000000
      protocol_version: 2.0

    wheels:
      separation: 0.287
      radius: 0.033

    motors:
      profile_acceleration_constant: 214.577

      # [rev/min2]
      # ref) http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#profile-acceleration
      profile_acceleration: 0.0

    sensors:
      bumper_1: 0
      bumper_2: 0
      illumination: 0
      ir: 0
      sonar: 0

diff_drive_controller:
  ros__parameters:

    odometry:
      publish_tf: true
      use_imu: true
      frame_id: "odom"
      child_frame_id: "base_footprint"

**robot.launch.py**
#!/usr/bin/env python3
#
# Copyright 2019 ROBOTIS CO., LTD.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
#
# Authors: Darby Lim

import os

from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration
from launch.substitutions import ThisLaunchFileDir
from launch_ros.actions import Node

def generate_launch_description():
    TURTLEBOT3_MODEL = os.environ['TURTLEBOT3_MODEL']
    LDS_MODEL = os.environ['LDS_MODEL']
    LDS_LAUNCH_FILE = '/hlds_laser.launch.py'

    usb_port = LaunchConfiguration('usb_port', default='/dev/ttyACM0')

    tb3_param_dir = LaunchConfiguration(
        'tb3_param_dir',
        default=os.path.join(
            get_package_share_directory('turtlebot3_bringup'),
            'param',
            TURTLEBOT3_MODEL + '.yaml'))

    if LDS_MODEL == 'LDS-01':
        lidar_pkg_dir = LaunchConfiguration(
            'lidar_pkg_dir',
            default=os.path.join(get_package_share_directory('hls_lfcd_lds_driver'), 'launch'))
    elif LDS_MODEL == 'LDS-02':
        lidar_pkg_dir = LaunchConfiguration(
            'lidar_pkg_dir',
            default=os.path.join(get_package_share_directory('ld08_driver'), 'launch'))
        LDS_LAUNCH_FILE = '/ld08.launch.py'
    else:
        lidar_pkg_dir = LaunchConfiguration(
            'lidar_pkg_dir',
            default=os.path.join(get_package_share_directory('hls_lfcd_lds_driver'), 'launch'))

    use_sim_time = LaunchConfiguration('use_sim_time', default='false')

    return LaunchDescription([
        DeclareLaunchArgument(
            'use_sim_time',
            default_value=use_sim_time,
            description='Use simulation (Gazebo) clock if true'),

        DeclareLaunchArgument(
            'usb_port',
            default_value=usb_port,
            description='Connected USB port with OpenCR'),

        DeclareLaunchArgument(
            'tb3_param_dir',
            default_value=tb3_param_dir,
            description='Full path to turtlebot3 parameter file to load'),

        IncludeLaunchDescription(
            PythonLaunchDescriptionSource(
                [ThisLaunchFileDir(), '/turtlebot3_state_publisher.launch.py']),
            launch_arguments={'use_sim_time': use_sim_time}.items(),
        ),

        IncludeLaunchDescription(
            PythonLaunchDescriptionSource([lidar_pkg_dir, LDS_LAUNCH_FILE]),
            launch_arguments={'port': '/dev/ttyUSB0', 'frame_id': 'base_scan'}.items(),
        ),

        Node(
            package='turtlebot3_node',
            executable='turtlebot3_ros',
            parameters=[tb3_param_dir],
            arguments=['-i', usb_port],
            output='screen'),
    ])

yigitboracagiran, 2024-08-09:
Can you share us the callback function?

Faheemibrahim, 2024-08-09:
i have added the content  in these files 

**launch and yaml files** 
[launch_file.txt](https://github.com/user-attachments/files/16558214/launch_file.txt)
[waffle_pi_yaml.txt](https://github.com/user-attachments/files/16558215/waffle_pi_yaml.txt)

**.cpp files** 
* for the output you have asked 
[odomentry_cpp.txt](https://github.com/user-attachments/files/16558221/odomentry_cpp.txt)
[joint_state_cpp.txt](https://github.com/user-attachments/files/16558224/joint_state_cpp.txt)

* i have added the rest just incase 
[diff_drive_controller_cpp.txt](https://github.com/user-attachments/files/16558228/diff_drive_controller_cpp.txt)
[dynamixel_sdk_wrapper_cpp.txt](https://github.com/user-attachments/files/16558229/dynamixel_sdk_wrapper_cpp.txt)
[imu_cpp.txt](https://github.com/user-attachments/files/16558230/imu_cpp.txt)
[node_main_cpp.txt](https://github.com/user-attachments/files/16558231/node_main_cpp.txt)
[sensor_state_cpp.txt](https://github.com/user-attachments/files/16558232/sensor_state_cpp.txt)
[turtlebot3_cpp.txt](https://github.com/user-attachments/files/16558233/turtlebot3_cpp.txt)

Please let me know if you need anything else, and I really appreciate your help !!

yigitboracagiran, 2024-08-09:
I didn't understand what you're trying to do with this. But did you try to download turtlebot3_simulation from [this address](https://github.com/ROBOTIS-GIT/turtlebot3_simulations) and launch it? Odom topic and your twist data most likely will come in simulation... After you complete this, we can go step by step...

yigitboracagiran, 2024-08-09:
Don't forget to put this repository to your workspace build with `colcon build`

Faheemibrahim, 2024-08-09:
I am working with a physical TurtleBot3 robot and have cloned the TurtleBot3 repository, specifically the release 2.1.5  https://github.com/ROBOTIS-GIT/turtlebot3/releases/tag/2.1.5  from GitHub, which is built for ROS 2 Humble. After cloning, I used colcon to build the workspace on the Raspberry Pi. To bring up the robot, I executed the following command on the Raspberry Pi  ros2 launch turtlebot3_bringup robot.launch.py

Then, from my laptop, I used the teleop package to control the robot. While the robot was moving, I monitored the odometry data by running: ros2 topic echo /odom but i dont see it updating the velocity values.

yigitboracagiran, 2024-08-09:
Did you check the frames?

Faheemibrahim, 2024-08-09:
[frames_2024-08-09_14.55.02.pdf](https://github.com/user-attachments/files/16560213/frames_2024-08-09_14.55.02.pdf)
yup heres an image

yigitboracagiran, 2024-08-09:
I didn't work on physical turtlebot3, do you still need to export to turtlebot3 model? Maybe you exported the wrong one?

Faheemibrahim, 2024-08-09:
No worries, I believe the turtlebot3 waffle_pi.Urdf is properly used when I launch the turtlebot_bringup. so I don’t think the issue is with it opening the wrong urdf. Could there be a problem with the urdf ?  The picture below shows the waffle_pi.urdf being used 
[image]

yigitboracagiran, 2024-08-09:
I see, what about coordinate information? Are they coming or not ( like velocity data ) with /odom topic? And what about other topics? For example is /cmd_vel coming true?

Faheemibrahim, 2024-08-09:
For /odom only the linear and angular velocity does not update but the coordinates update normally. And /cmd_vel is updating normally 
[image]

Faheemibrahim, 2024-08-09:
The topic updates the the Linear twist and angular twist but the covariance is not updating. Will that be an issue if I use slam or nav2 ? and yes the /cmd_vel updates perfectly 
[image]

yigitboracagiran, 2024-08-09:
Is /odom topic successfully publish the points?

Faheemibrahim, 2024-08-09:
[image]
i zoomed out and found this is this how it should be updating ?

yigitboracagiran, 2024-08-09:
Can you run `ros2 topic echo /odom --no-arr` in a new terminal next to your bringup and teleop nodes?
You can see the robot coordinates in the terminal?
Does it coming true or not?

Faheemibrahim, 2024-08-09:
The coordinates are being published correctly but for some reason it starts at x =26 and y = 14 and the z updates properly.

yigitboracagiran, 2024-08-09:
Can you reboot the robot?
If it still  starts at (26, 14) probably there are initialize values in a launch file. Can you check please?

Faheemibrahim, 2024-08-09:
Rebooting it didn’t work physically turning it off and rebooting as you suggested worked. So is the odom fine ?

yigitboracagiran, 2024-08-09:
Yes it's looks like it

yigitboracagiran, 2024-08-09:
If odom coordinates and cmd_vel topics are coming true, linear and angular velocity in /odom should be fine also. Can't you use /cmd_vel in your code instead of using velocities in /odom for now until we figuring it out why the velocity data is coming wrong in /odom?

Faheemibrahim, 2024-08-09:
Thank you for looking into this issue and being patient with me. may god bless you bro !!

Faheemibrahim, 2024-08-09:
I thought odom data publishes everything. i just starting learning ros and was trying to make my own control system to experiment and I thought it was important for the odom to publish velocities and thought it wasn’t working.

Faheemibrahim, 2024-08-09:
So odom only publishes coordinates ? And I use cmd_vel to track the velocity ?

yigitboracagiran, 2024-08-09:
Exactly, best way to track velocity is subscribing velocity topic, in our case subscribing /cmd_vel

Faheemibrahim, 2024-08-09:
Honestly man you’re the best thank you !!

closed 2024-08-09 by Faheemibrahim