ROBOTIS-GIT/turtlebot3 #1030: ros2 foxy turtlbot3 burger sbc setup error with stderr

ankita-kalra, 2024-08-30:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [0] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [0] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [0] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [0] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [0] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    - Summary: 7 packages finished [2min 19s] 1 package had stderr output: ld08_driver
    stderr: ld08_driver
/home/ubuntu/turtlebot3_ws/src/ld08_driver/src/transform.cpp: In constructor 'SlTransform:SlTransform(LDVersion, bool)':
/home/ubuntu/turtlebot3_ws/src/ld08_driver/src/transform.cpp:30:50: warning: unused parameter 'to_right_hand' [-Wunused-parameter]
30 | SlTransform::SlTransform(LDVersion version, bool to_right_hand)

yun-goon, 2024-10-31:
Hello, @ankita-kalra 

This message is a warning, not an error, and it does not affect functionality.
You can proceed without concern.

Thank you.

sunghowoo, 2024-12-06:
This issue appears to be resolved, so it will be closed.

closed 2024-12-06 by sunghowoo