ROBOTIS-GIT/turtlebot3 #1033: Namespace

faberpis, 2024-10-14:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [x] etc (Please specify your ROS Version here)
ROS2 Humble
3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)
5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [x] etc (Please specify your OS here)
Ubuntu 22.04
6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

8. Copy and Paste the error messages on terminal.
Fixed frame [map] does not exists

9. Please describe the issue in detail.
I am working with a real Turtlebot3 Burger and need to operate a multi-agent SLAM. I am having issues with namespacing, does anyone know what changes need to be made?

GyuH13, 2024-11-06:
Hi @faberpis
https://github.com/ROBOTIS-GIT/turtlebot3/issues/1018#issuecomment-2458631984
Refer this for setting namespace.

sunghowoo, 2025-01-13:
@faberpis Please refer to GyuH13's answer for initial guidance. This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it, and we will address it promptly.

closed 2025-01-13 by sunghowoo