ROBOTIS-GIT/turtlebot3 #1034: Cannot trigger OpenCR sounds when publishing to `/sound` topic (ROS2 Humble)

tom-howard, 2024-11-13:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [x] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [x] ROS 2 Humble Hawksbill

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [x] [UP Squared](https://up-shop.org/default/up-squared-series.html)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [x] Ubuntu 22.04 Server

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [x] Ubuntu 22.04 LTS (Jammy Jellyfish)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: humble
    - Firmware version: latest ROS2 version

7. Specify the commands or instructions to reproduce the issue.

    (All running on the SBC):
    
    1. Launching the robot bringup:

        ```
        ros2 launch turtlebot3_bringup robot.launch.py
        ```
    1. Publishing messages to `/sound` to trigger sounds from the OpenCR:

        ```
        ros2 topic pub /sound turtlebot3_msgs/msg/Sound "value: 1"
        ```

8. Copy and Paste the error messages on terminal.

    - No error messages, just nothing happens

9. Please describe the issue in detail.

    On ROS1 Noetic it was possible to publish messages to the `/sound` topic to make the OpenCR play a handful of pre-configured sounds. This no longer seems to work with the same robot running ROS2 (Humble), and with the latest ROS2 firmware installed on the OpenCR. Sending messages to the topic (e.g. `ros2 topic pub /sound turtlebot3_msgs/msg/Sound "value: 1"`) seem to have no effect.

    On launching the robot bringup, I get a log message from the `turtlebot3_node` saying `"Succeeded to create sound server"` and I've confirmed that the topic is visible from a separate terminal when I run `ros2 topic list`. From a separate terminal I can echo the messages that I'm publishing on `/sound` (`ros2 topic echo /sound`) and can confirm that the messages *are* being published to the topic correctly. Other functionality on the robot works fine (teleoperation etc) and the OpenCR **does** play the 'doremi' noise on start up, therefore the buzzer is definitely working. 

    Has this functionality been removed? Or am I doing something wrong?

yun-goon, 2024-11-14:
Hello @tom-howard 

In the turtlebot3_node/src/devices/sound.cpp file, you'll notice that sound is managed as a service. After running the bringup on the TurtleBot, you can check if sound is working by entering the following command in the terminal:
```
ros2 service call /sound turtlebot3_msgs/srv/Sound "{value: 1}"
```
Please leave a comment if you encounter any other issues. 
Thank you!

tom-howard, 2024-11-14:
Ah, that works, thank you!

closed 2024-11-14 by tom-howard