ROBOTIS-GIT/turtlebot3 #1035: ros2 unable to find ~/map.yaml when attempting to simulate navigation on host pc

AbduEhab, 2024-11-15:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [x] ROS 2 Humble

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [x] Pc Gazebo sim

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [x] Pc Gazebo sim

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [x] Ubuntu 22.04 LTS

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - cd ~/turtlebot3_ws/ && source install/local_setup.bash && export TURTLEBOT3_MODEL=waffle_pi && ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
    - cd ~/turtlebot3_ws/ && source install/local_setup.bash && export TURTLEBOT3_MODEL=waffle_pi && ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True map:=$HOME/map.yaml

8. Copy and Paste the error messages on terminal.

    [image]

9. Please describe the issue in detail.

    - I can't load the map.yaml that I generated in the SLAM section of the turtlebot3 docs.
    - I made sure that I actually had the files.

[image]

All the issues and solutions I found were in relation to either ROS1 or the navigation on the PI from a remote PC. But there was nothing that I could find for local sims.

Here is a screenshot of the issue:
[image]

AbduEhab, 2024-11-15:
To clarify. I am not connected to any turtlebots. I want to simulate something on my local machine.

yun-goon, 2024-11-18:
Hello @AbduEhab 

The tf issue is something we’re aware of and actively working to improve. However, that error shouldn’t prevent the navigation2 launch from functioning correctly.

Looking at your RViz image, it seems the issue is similar to what happens when the map file isn’t actually located in the $HOME directory. Here's a solution to try:

First, launch without specifying a map:
Run the following command:

```
ros2 launch turtlebot3_manipulation_navigation2 navigation2.launch.py
```
If you don’t specify a map, the package should load a default map stored internally.

Next, confirm if your map file exists in $HOME:
Check if your map file is actually present in the $HOME directory. If it’s missing, you can follow the instructions in the TurtleBot3 e-manual to generate one:
[TurtleBot3 Simulation Manual](https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/).

If the problem remains even after generating the map and placing it in $HOME, please let me know so we can investigate further.

Thanks!

AbduEhab, 2024-11-18:
Hello,

When I tried to launch using your command. I was given the same error.
[image]

I have also made sure that the maps are in the $HOME dir
[image]

If it is missing the default map, how can I fix that?

yun-goon, 2024-11-19:
Hello,

I’m sorry to hear that the issue couldn’t be resolved.
Normally, it should work without issues, but this could happen if the files were improperly installed or if there are dependency issues.

Could you try the following steps to address this?

Refer to the [TurtleBot3 manual](https://emanual.robotis.com/docs/en/platform/turtlebot3/overview/) and attempt a clean reinstallation.

Run the following commands to resolve dependency issues automatically:

```
cd ~/(your_workspace_folder)
sudo rosdep init
rosdep update
rosdep install --from-paths src --ignore-src -r -y
```
Additionally, you can try installing all navigation-related packages using this command:

```
sudo apt install ros-humble-nav*
After following these steps, check if the issue persists.
```
Lastly, I have a quick question: Are you running this on a virtual machine?

I’ll keep investigating the root cause of this issue. Let me know if you find anything new or if there’s additional information that might help resolve it!

GyuH13, 2025-02-19:
Hi @AbduEhab 
Have you resolved the issue?
Since there are no further questions, I will be closing this issue soon.
If the issue has not been resolved yet, please let me know. I’ll be happy to assist you further.

AbduEhab, 2025-02-20:
> Run the following commands to resolve dependency issues automatically:
> 
> ```
> cd ~/(your_workspace_folder)
> sudo rosdep init
> rosdep update
> rosdep install --from-paths src --ignore-src -r -y
> ```

Hello, sorry I was away for a while.

Running these few lines seems to have fixed the missing map issue, but now the map does not show on the screen. Any idea why that could be the case?

I am running it on a KVM. Do I need to do something different in this case as if I were to run it on direct hardware?

Thank you for your help.

GyuH13, 2025-02-24:
This issue has been resolved, so I will close it. Thank you!

GyuH13, 2025-02-24:
Does below error appear when you start the navigation?
[image]
I'm trying to reproduce your situation, but I can only recreate the same issue when the map file path is incorrect.
The issue might be caused by KVM preventing the file path from being found. How about trying an absolute path?

AbduEhab, 2025-02-24:
No the error does not show.

I did an installation on a laptop instead of a KVM and the same problem still persists. Here are the errors that show.

[image]

[image]

khinggan, 2025-05-15:
Hi @AbduEhab, 

I met the same problem when testing after building the source code from an older version of the humble branch.

I was able to fix the issue by checking the version history of the [waffle_pi.yaml](https://github.com/ROBOTIS-GIT/turtlebot3/commits/humble/turtlebot3_navigation2/param/waffle_pi.yaml) file.

In the commit [568bcc3](https://github.com/ROBOTIS-GIT/turtlebot3/commit/568bcc35e887fca367d2a58fb4addad39abedf9e) from October 30, 2022, the `scan_topic: scan` parameter is missing in `amcl: ros_parameters`.

Best regards,

GyuH13, 2025-05-15:
@AbduEhab 
Did you manage to resolve the issue?
[image]
From the terminal output, it looks like the map was loaded correctly, but the AMCL node didn’t start properly. Based on the error message in the terminal, it seems the problem is related to the robot model type specified in the parameters.

@khinggan 
Hello.
Thanks for finding missing.
I will fix it when I update tb3_nav package about launch.
Since the AMCL node uses the scan topic as a default parameter, I think that may not be the cause.