ROBOTIS-GIT/turtlebot3 #1037: Velocities reported by /odom topic on a real robot are extremely low

marco-ambrosio, 2024-11-20:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ x] ROS 2 Humble

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [x] Ubuntu 22.04 LTS
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - Run the bring up
    - Use the turtlebot3_teleop to send velocity commands
    - echo the /odom topic and look at twist

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    - The issue is described and solved in [#1001](https://github.com/ROBOTIS-GIT/turtlebot3/pull/1001)
    - please merge the solution :)

sunghowoo, 2024-12-06:
@marco-ambrosio Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues and are actively working on resolutions. We are currently testing your pull request. Please bear with us, and thank you for your patience and contribution!

sunghowoo, 2025-01-08:
https://github.com/ROBOTIS-GIT/turtlebot3/pull/1001 was successfully merged.

closed 2025-01-08 by sunghowoo