ROBOTIS-GIT/turtlebot3 #1038: [Notice] Nav2 Migrated to TwistStamped to replace Twist for cmd_vel Topics

SteveMacenski, 2024-12-03:
Hi,

As a key community project using Nav2, I wanted to drop you a note that I am in the process of changing the default cmd_vel topic type from Twist to TwistStamped in efforts of enabling a broader set of applications and safety constraints. This has been an on-going discussion over the last 18 months and now has the technology stack in GZ, ROS Control, and Nav2 to enable it.

To migrate, your robot and simulation should soon accept a TwistStamped message rather than a Twist message. We have a couple of provided utilities that could make this easier for you, by switching to use a Twist Publisher and Twist Subscriber [1-2]. This is what is now used in all Nav2 nodes to either publish or subscribe to a Twist or TwistStamped message.

If you would like to continue to use a Twist, you can set enable_stamped_cmd_vel to false in all nodes in your configuration that use Twist Publisher/Subscriber objects, but this will be shortly switched to being default true.

[1] https://github.com/ros-navigation/navigation2/blob/main/nav2_util/include/nav2_util/twist_publisher.hpp
[2] https://github.com/ros-navigation/navigation2/blob/main/nav2_util/include/nav2_util/twist_subscriber.hpp

Nav2 migration:
- https://github.com/ros-navigation/navigation2/pull/4779
- https://github.com/ros-navigation/docs.nav2.org/pull/614
- https://github.com/ros-navigation/nav2_minimal_turtlebot_simulation/pull/16

robotpilot, 2024-12-03:
Hi @SteveMacenski 
I support the changing the default cmd_vel topic type from Twist to TwistStamped in the Nav2 community and will be migrating to this change in TurtleBot3.

SteveMacenski, 2024-12-03:
BTW: This is for Kilted / Rolling by default, but can be enabled in Jazzy as well (for a current migration, if you choose) by toggling the `enable_stamped_cmd_vel: true` parameter for each of the Nav2 nodes

robotpilot, 2024-12-03:
Okay! Thanks, @SteveMacenski

GyuH13, 2025-04-01:
Hello, @SteveMacenski 
Starting from the TurtleBot3 2.2.6 release, the Jazzy version supports the `/cmd_vel` of type TwistStamped.
From the Jazzy version onwards, `/cmd_vel` of type TwistStamped is used by default.
Additionally, if you want to use the Twist type, you can toggle it by setting `enable_stamped_cmd_vel = false`, just like in Nav2. (The default value is true)

Thank you for the notice.I will close this issue.

closed 2025-04-01 by GyuH13