ROBOTIS-GIT/turtlebot3 #1042: [Humble] turtlebot3_manipulation_bringup Error; [ros2_control_node-1]: process has died

LUGGGI, 2025-01-04:
ISSUE TEMPLATE ver. 0.4.2

**Please search for similar issues in the repository before submit an issue ticket.**

1. Select the TurtleBot3 model.

    - [x ] Waffle Pi with OpenMANIPULATOR

2. Which SBC(Single Board Computer) is installed on TurtleBot3?

    - [x ] Raspberry Pi 4 (specify RAM size: 2/4/8GB)

3. Which OS is installed on the TurtleBot3 SBC?

    - [x] Ubuntu preinstalled server 22.04

4. Which OS is installed on Remote PC?

    - [x] Ubuntu 22.04 LTS (Jammy Jellyfish)

5. Which ROS version do you use?

    - [x] ROS2 Humble Hawksbill

6. Please describe the issue.

Wenn runnig the bringup for the manipulation the ros2_control_node dies because of an error.
`ros2 launch turtlebot3_manipulation_bringup hardware.launch.py`

The line bevor shows the error `[ros2_control_node-1] *** stack smashing detected ***: terminated` also the error `[ros2_control_node-1] [ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet!` shows iü

7. Copy & paste error messages printed on the screen. Screenshots or video clips are welcome.

```
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-01-04-11-58-16-065808-ubuntu-1069
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [1072]
[INFO] [robot_state_publisher-2]: process started with pid [1074]
[INFO] [spawner-3]: process started with pid [1076]
[INFO] [ld08_driver-4]: process started with pid [1078]
[robot_state_publisher-2] [INFO] [1735988298.397978923] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-2] [INFO] [1735988298.398583851] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1735988298.398729318] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-2] [INFO] [1735988298.398768412] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-2] [INFO] [1735988298.398796524] [robot_state_publisher]: got segment camera_rgb_frame
[robot_state_publisher-2] [INFO] [1735988298.398823747] [robot_state_publisher]: got segment camera_rgb_optical_frame
[robot_state_publisher-2] [INFO] [1735988298.398852471] [robot_state_publisher]: got segment caster_back_left_link
[robot_state_publisher-2] [INFO] [1735988298.398878323] [robot_state_publisher]: got segment caster_back_right_link
[robot_state_publisher-2] [INFO] [1735988298.398903843] [robot_state_publisher]: got segment dummy_mimic_fix
[robot_state_publisher-2] [INFO] [1735988298.398929381] [robot_state_publisher]: got segment end_effector_link
[robot_state_publisher-2] [INFO] [1735988298.398956141] [robot_state_publisher]: got segment gripper_left_link
[robot_state_publisher-2] [INFO] [1735988298.398981864] [robot_state_publisher]: got segment gripper_right_link
[robot_state_publisher-2] [INFO] [1735988298.399007346] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-2] [INFO] [1735988298.399033143] [robot_state_publisher]: got segment link1
[robot_state_publisher-2] [INFO] [1735988298.399058570] [robot_state_publisher]: got segment link2
[robot_state_publisher-2] [INFO] [1735988298.399084071] [robot_state_publisher]: got segment link3
[robot_state_publisher-2] [INFO] [1735988298.399109109] [robot_state_publisher]: got segment link4
[robot_state_publisher-2] [INFO] [1735988298.399134443] [robot_state_publisher]: got segment link5
[robot_state_publisher-2] [INFO] [1735988298.399159221] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-2] [INFO] [1735988298.399184907] [robot_state_publisher]: got segment wheel_right_link
[ros2_control_node-1] [WARN] [1735988298.540418103] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-1] [INFO] [1735988298.542210365] [resource_manager]: Loading hardware 'TurtleBot3ManipulationSystem'
[ros2_control_node-1] [INFO] [1735988298.564666918] [resource_manager]: Initialize hardware 'TurtleBot3ManipulationSystem'
[ros2_control_node-1] [INFO] [1735988298.568268425] [turtlebot3_manipulation]: Succeeded to open port
[ros2_control_node-1] [INFO] [1735988298.575950214] [turtlebot3_manipulation]: Succeeded to set baudrate
[ros2_control_node-1] [INFO] [1735988298.583886438] [turtlebot3_manipulation]: OpenCR Model Number 20480
[ros2_control_node-1] [INFO] [1735988298.617329240] [turtlebot3_manipulation]: Connected manipulator
[ros2_control_node-1] [INFO] [1735988298.650369993] [turtlebot3_manipulation]: Connected wheels
[ros2_control_node-1] [INFO] [1735988298.650730801] [resource_manager]: Successful initialization of hardware 'TurtleBot3ManipulationSystem'
[ros2_control_node-1] [INFO] [1735988298.651423490] [resource_manager]: 'configure' hardware 'TurtleBot3ManipulationSystem'
[ros2_control_node-1] [INFO] [1735988298.651500937] [resource_manager]: Successful 'configure' of hardware 'TurtleBot3ManipulationSystem'
[ros2_control_node-1] [INFO] [1735988298.651539179] [resource_manager]: 'activate' hardware 'TurtleBot3ManipulationSystem'
[ros2_control_node-1] [INFO] [1735988298.651561624] [turtlebot3_manipulation]: Ready for start
[ros2_control_node-1] [INFO] [1735988298.684292200] [turtlebot3_manipulation]: Wait for IMU re-calibration
[ros2_control_node-1] [ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet!
[ld08_driver-4] /dev/ttyACM0    OpenCR Virtual ComPort in FS Mode
[ld08_driver-4] Can't find LDS-02
[INFO] [ld08_driver-4]: process has finished cleanly [pid 1078]
[spawner-3] [INFO] [1735988299.728820512] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[ros2_control_node-1] [INFO] [1735988301.718971073] [turtlebot3_manipulation]: Joints and wheels torque ON
[ros2_control_node-1] [INFO] [1735988301.807787292] [turtlebot3_manipulation]: Set profile acceleration and velocity to joints
[ros2_control_node-1] [INFO] [1735988301.942852263] [turtlebot3_manipulation]: Set profile acceleration and velocity to gripper
[ros2_control_node-1] [INFO] [1735988302.078859989] [turtlebot3_manipulation]: Set goal current value to gripper
[ros2_control_node-1] [ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet!
[ros2_control_node-1] [INFO] [1735988302.147185463] [turtlebot3_manipulation]: System starting
[ros2_control_node-1] [INFO] [1735988302.147897426] [resource_manager]: Successful 'activate' of hardware 'TurtleBot3ManipulationSystem'
[ros2_control_node-1] [INFO] [1735988302.176412446] [controller_manager]: update rate is 100 Hz
[ros2_control_node-1] [INFO] [1735988302.176526430] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-1] [WARN] [1735988302.177018646] [controller_manager]: No real-time kernel detected on this system. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[ros2_control_node-1] [INFO] [1735988302.177260986] [turtlebot3_manipulation]: Start to read wheels and manipulator states
[ros2_control_node-1] [INFO] [1735988302.181068545] [turtlebot3_manipulation]: Start to write wheels and manipulator commands
[ros2_control_node-1] *** stack smashing detected ***: terminated
[ERROR] [ros2_control_node-1]: process has died [pid 1072, exit code -11, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_7tr19tsr --params-file /home/ubuntu/turtlebot3_ws/install/turtlebot3_manipulation_bringup/share/turtlebot3_manipulation_bringup/config/hardware_controller_manager.yaml -r ~/cmd_vel_unstamped:=cmd_vel -r ~/odom:=odom'].
[spawner-3] [WARN] [1735988312.241116969] [spawner_joint_state_broadcaster]: Failed getting a result from calling /controller_manager/list_controllers in 10.0. (Attempt 1 of 3.)
```

8. How to reproduce the issue?

Install the SBC and OpenRC as described in the [Manual](https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/)

LUGGGI, 2025-01-04:
Things I have already tried:

- The `[ld08_driver-4] Can't find LDS-02` is not the reason, same error if it is not unpled
- With the setup for the standalown [openmanipulator_x](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide/) it works as expected
- With the ROS1 Noetic install it also works (but i need ros2 humble)
- Flashing the OpenCR with ArduinoIDE (and older versions of the library) doesn't work

LUGGGI, 2025-01-04:
After updating the dynamixels the manipulator went into the home position but also hat the same error but different log messages:

```
1736009454.2483506 [INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-01-04-17-50-54-242432-ubuntu-1083
1736009454.2490804 [INFO] [launch]: Default logging verbosity is set to INFO
1736009456.0678923 [INFO] [ros2_control_node-1]: process started with pid [1086]
1736009456.0689926 [INFO] [robot_state_publisher-2]: process started with pid [1088]
1736009456.0697713 [INFO] [spawner-3]: process started with pid [1090]
1736009456.0704522 [INFO] [ld08_driver-4]: process started with pid [1092]
1736009456.7136934 [ros2_control_node-1] [WARN] [1736009456.712784982] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
1736009456.7159669 [ros2_control_node-1] [INFO] [1736009456.715100954] [resource_manager]: Loading hardware 'TurtleBot3ManipulationSystem' 
1736009456.7401114 [ros2_control_node-1] [INFO] [1736009456.739329950] [resource_manager]: Initialize hardware 'TurtleBot3ManipulationSystem' 
1736009456.7464678 [ros2_control_node-1] [INFO] [1736009456.745647869] [turtlebot3_manipulation]: Succeeded to open port
1736009456.7559450 [ros2_control_node-1] [INFO] [1736009456.755315460] [turtlebot3_manipulation]: Succeeded to set baudrate
1736009456.7715225 [ros2_control_node-1] [INFO] [1736009456.770707250] [turtlebot3_manipulation]: OpenCR Model Number 20480
1736009456.8053427 [ros2_control_node-1] [INFO] [1736009456.804681281] [turtlebot3_manipulation]: Connected manipulator
1736009456.8414745 [ros2_control_node-1] [INFO] [1736009456.840685971] [turtlebot3_manipulation]: Connected wheels
1736009456.8432574 [ros2_control_node-1] [INFO] [1736009456.841057562] [resource_manager]: Successful initialization of hardware 'TurtleBot3ManipulationSystem'
1736009456.8438492 [ros2_control_node-1] [INFO] [1736009456.841749652] [resource_manager]: 'configure' hardware 'TurtleBot3ManipulationSystem' 
1736009456.8442752 [ros2_control_node-1] [INFO] [1736009456.841829281] [resource_manager]: Successful 'configure' of hardware 'TurtleBot3ManipulationSystem'
1736009456.8446746 [ros2_control_node-1] [INFO] [1736009456.841872910] [resource_manager]: 'activate' hardware 'TurtleBot3ManipulationSystem' 
1736009456.8450909 [ros2_control_node-1] [INFO] [1736009456.841896632] [turtlebot3_manipulation]: Ready for start
1736009456.8763890 [ros2_control_node-1] [INFO] [1736009456.875666275] [turtlebot3_manipulation]: Wait for IMU re-calibration
1736009456.9107149 [ros2_control_node-1] [1;31m[ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet![0m
1736009457.3150735 [INFO] [ld08_driver-4]: process has finished cleanly [pid 1092]
1736009459.9117384 [ros2_control_node-1] [INFO] [1736009459.910412353] [turtlebot3_manipulation]: Joints and wheels torque ON
1736009459.9581766 [ros2_control_node-1] [1;31m[ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet![0m
1736009459.9599562 [ros2_control_node-1] [INFO] [1736009459.959096455] [turtlebot3_manipulation]: Set profile acceleration and velocity to joints
1736009459.9943533 [ros2_control_node-1] [1;31m[ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet![0m
1736009460.0290685 [ros2_control_node-1] [1;31m[ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet![0m
1736009460.0306528 [ros2_control_node-1] [INFO] [1736009460.029121175] [turtlebot3_manipulation]: Set profile acceleration and velocity to gripper
1736009460.0640893 [ros2_control_node-1] [1;31m[ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet![0m
1736009460.0999434 [ros2_control_node-1] [1;31m[ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet![0m
1736009460.1015077 [ros2_control_node-1] [INFO] [1736009460.100129540] [turtlebot3_manipulation]: Set goal current value to gripper
1736009460.1351862 [ros2_control_node-1] [1;31m[ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet![0m
1736009460.1367185 [ros2_control_node-1] [INFO] [1736009460.135104252] [turtlebot3_manipulation]: System starting
1736009460.1700938 [ros2_control_node-1] [1;31m[ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet![0m
1736009460.1707375 [ros2_control_node-1] [INFO] [1736009460.169445078] [resource_manager]: Successful 'activate' of hardware 'TurtleBot3ManipulationSystem'
1736009460.1983168 [ros2_control_node-1] [INFO] [1736009460.197536002] [controller_manager]: update rate is 100 Hz
1736009460.1991713 [ros2_control_node-1] [INFO] [1736009460.197653576] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
1736009460.2005131 [ros2_control_node-1] [WARN] [1736009460.198094796] [controller_manager]: No real-time kernel detected on this system. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
1736009460.2009950 [ros2_control_node-1] [INFO] [1736009460.198276851] [turtlebot3_manipulation]: Start to read wheels and manipulator states
1736009460.2079651 [ros2_control_node-1] [INFO] [1736009460.207311445] [turtlebot3_manipulation]: Start to write wheels and manipulator commands
1736009460.2424827 [ros2_control_node-1] [1;31m[ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet![0m
[... cut ...]
1736009460.4225807 [ros2_control_node-1] *** stack smashing detected ***: terminated
1736009461.8510973 [ERROR] [ros2_control_node-1]: process has died [pid 1086, exit code -11, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_y0avqhl6 --params-file /home/ubuntu/turtlebot3_ws/install/turtlebot3_manipulation_bringup/share/turtlebot3_manipulation_bringup/config/hardware_controller_manager.yaml -r ~/cmd_vel_unstamped:=cmd_vel -r ~/odom:=odom'].
```

yun-goon, 2025-01-06:
Hello @LUGGGI 
Thanks to you, I realized there was an issue with the manual, and I'm currently working on fixing it.
Also, I'm investigating the issue. For now, it's okay if the [ERROR] [DynamixelSDKWrapper] [TxRxResult] There is no status packet! error appears once. I'm looking into the cause and addressing it.
I'm trying to reproduce the issue and resolve it step by step.
In the meantime, could you refer to the manual and try setting up the SBC and OpenCR again?

LUGGGI, 2025-01-06:
Hello @yun-goon,
thanks for the reply.

> Hello @LUGGGI Thanks to you, I realized there was an issue with the manual, and I'm currently working on fixing it. 

Do you mean that? ROBOTIS-GIT/emanual#1729
Also noticed another problem with the setup: ROBOTIS-GIT/emanual#1731

> In the meantime, could you refer to the manual and try setting up the SBC and OpenCR again?

I did that, I still get the same errors as in the second log from my previous comment (I also just realised that I copied the wrong one in, fixed that now).

@yun-goon if there is anything more i could try or help you reproduce the issue please ask.

sunghowoo, 2025-01-08:
@LUGGGI This issue might occurs with the latest version of Dynamixel, and it has been confirmed that it can be resolved by using the Dynamixel Wizard to change the **Return Delay Time** of all motors from 250 ms to **0 ms.**

LUGGGI, 2025-01-08:
@sunghowoo Thank you very much. I just tried your suggestion and it works for me. 

I still get the first `ros2 run turtlebot3_manipulation_teleop turtlebot3_manipulation_teleop' error, but that doesn't seem to be a problem.

closed 2025-01-13 by sunghowoo