ROBOTIS-GIT/turtlebot3 #1044: multiple maps in SLAM & wrong orientation in Navigation

AmirrezaRamesh, 2025-01-08:
1. Select the TurtleBot3 model.

    - [ ] Burger
    - [x] Waffle Pi

2. Which SBC(Single Board Computer) is installed on TurtleBot3?

    - [x] Raspberry Pi 3 B+
    - [ ] Raspberry Pi 4 (specify RAM size: 2/4/8GB)
    - [ ] other (specify the product)

3. Which OS is installed on the TurtleBot3 SBC?

    - [ ] RPi Recovery image by ROBOTIS
    - [ ] Ubuntu preinstalled server 18.04
    - [ ] Ubuntu preinstalled server 20.04
    - [x] Ubuntu preinstalled server 22.04
    - [ ] other (specify the OS)

4. Which OS is installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [x] Ubuntu 22.04 LTS (Jammy Jellyfish)
    - [ ] other (specify the OS)

5. Which ROS version do you use?

    - [ ] ROS Kinetic Kame (EOL)
    - [ ] ROS2 Dashing Diademata (EOL)
    - [ ] ROS2 Eloquent Elusor (EOL)
    - [ ] ROS Melodic Morenia
    - [ ] ROS Noetic Ninjemys
    - [ ] ROS2 Foxy Fitzroy
    - [ ] ROS2 Galactic Geochelone
    - [x] ROS2 Humble Hawksbill
    - [ ] ROS2 Rolling Ridley

6. Please describe the issues.
 
- when running **SLAM node** using `ros2 launch turtlebot3_cartographer cartographer.launch.py` , there are multiple maps made unless we move the robot super slow! as soon as the robot goes a straight line with a normal speed or simply turns,you will see new walls in the map a little bit closer or further away from the original walls first detected by Lidar.

- when using SLAM node, **odom frame** constantly moves and changes its place to a random place.

- after saving a map and running the **turtlebot nav2 node**, the map is not the same as the one that pops when using the same commands for the **turtlebot gazebo simulator node**. for example in the one I test on the real robot, there is no **global planner** or **local planner**. to express the diffrence better, when using nav2 for simulator, the background of map becomes purle and there is a square around the robot in which the area is transparent.

- when launching the robot, and running `rqt_graph` we well see a node publishing to /scan topic but after launching Rviz2, and using `rqt_graph` there is no nodes shown to be publishing to /scan topic yet checking the info of the topic, it says there is a publisher, but still when choosing the /scan topic in Rviz2, no live data of Lidar is shown(which is expected to be shown in red dots). but that doesn't mean we don't actually receive the data because when using cartographer, the mapping gets done by the Lidar sensor.

- even if we manage to get a map by moving the robot super slowly, when using the **turtlebot nav2 node** ,  the is a **colorful static data of lidar** which appears after first detemining the initial position of the robot. but as the robot moves, this **colorful Lidar data** detects the same objects and walls in an off-position which makes the map so messy and when the robot goal is to go to a corner of the map and stand in a specific orientation, the robot hardly manages to reach the point and can't find the right orientation and constantly rotates around itself in a way which seems like the robot hasn't fully figured out his position, either because of /odom or /scan or etc

[image]

 
I will be super thankful if anyone could help me out with this <3

GyuH13, 2025-02-19:
Hi @AmirrezaRamesh 
The issue of the odom frame moving to a random position can occur if there are multiple topics using the /odom name in the same network. Are you using multiple TurtleBots on the same network?

If so, all the problems with Nav2 and SLAM that you're experiencing are likely caused by the changes in the odom frame.

Please check how many nodes are publishing to the /odom topic by running ros2 topic info /odom.

AmirrezaRamesh, 2025-04-14:
thanks for the reply;
after undoing the changes I'd made and a fresh reinstallation, most of problems seems to have gone away. I haven't had the time to get fully into it ,but still there are a few minor problems worth mentioning. I will be reporting what you asked me very soon!

GyuH13, 2025-04-15:
Hello, @AmirrezaRamesh 
I'm glad to hear that the issue has been partially resolved.
TurtleBot3 is currently undergoing continuous updates with many improvements being made.
We recommend pulling the latest version of the branch you are using to take advantage of the most up-to-date TurtleBot3 packages.
If you encounter any further issues or have additional questions, please feel free to let us know with as many details as possible.
We'll do our best to support you. 
Thank you!