ROBOTIS-GIT/turtlebot3 #1048: [ISSUE] No map received warning in RVIZ2 when following navigation example

PVBrodersen, 2025-01-23:
Hi,

I am currently testing the nav2 stack on a physical TB3, and have gotten everything up and running till the point where I want to load my map into RVIZ2 -- follow [this example](https://emanual.robotis.com/docs/en/platform/turtlebot3/navigation/#run-navigation-nodes). However, when I try to launch everything, I am met with the following prompt in RVIZ2. 

[image]

As seen, the occupancy grid generated from the previous SLAM example (ensured that everything was called/saved correctly) is not shown in RVIZ2, and it promptly says that no map is received. Hereafter, I checked the TF2 view frames graph, as the logs from my terminal complained about a missing transform -- prompt:

[image]

Resulting in this TF2 frame graph: 
[frames_2024-05-03_13.36.35.pdf](https://github.com/ROBOTIS-GIT/emanual/files/15199487/frames_2024-05-03_13.36.35.pdf)

I am a bit unsure how to proceed, and any help would be greatly appreciated :)

GyuH13, 2025-02-04:
Hello @PVBrodersen 
Sorry for the late response.
According to the provided Rviz, the map frame does not exist.
If the fixed frame is set to map, and the map created by SLAM does not appear in Rviz2, please double-check the map argument. Ensure that the correct YAML file path is specified.

If you have any additional issues, please feel free to ask!

GyuH13, 2025-02-19:
@PVBrodersen 
There are no further questions, so I will close this issue. If you have any additional questions, feel free to reopen the issue. I'll do my best to assist you. Thank you!

closed 2025-02-19 by GyuH13