ROBOTIS-GIT/turtlebot3 #1049: No robot model or localisation (TF warnings) when running cartographer (in simulation)

tom-howard, 2025-01-30:
## Context 
1. **TurtleBot3 model**: waffle
1. **ROS Version**: Humble on Ubuntu 22.04 LTS (Jammy Jellyfish) 

## Issue
I'm having errors when running SLAM in a simulation environment (e.g. "Turtlebot3 World") where cartographer seems to build a map, but there is no localisation, and I get a series of TF errors in RViz.

**Step 1 - Launch the sim**:
```bash
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
```

**Step 2 - Launch cartographer**: 
```bash
ros2 launch turtlebot3_cartographer cartographer.launch.py
```

As shown in the screenshot below, there are warnings for all model links under `TF` in RViz stating: `No transform from [XXX] to [map]`. 

[image]

This does not seem to happen when I run cartographer with a real robot in a physical environment.

Additionally, with *Gmapping in Noetic* the robot model used to be displayed, e.g.:

[image]

... but this no longer seems to be the case in Humble, both for real world applications nor in simulation ([see here for example](https://tom-howard.github.io/com2009/waffles/basics/#slam-steps)).

Please help!

Thanks,
Tom

GyuH13, 2025-01-31:
Hi @tom-howard ,
I checked this issue and tested it with the frame graph.
The TF between the map frame and other frames is being published normally.
The reason why RViz shows the "No transform from [XXX] to [map]" error is that the timestamps are not synchronized.
With the command above, I was able to run Cartographer normally.

```bash
 ros2 launch turtlebot3_cartographer cartographer.launch.py use_sim_time:=true

```

[image]  

If you have any additional questions, feel free to ask.

sunghowoo, 2025-01-31:
@tom-howard Hello Dr. Howard, thank you for your report. This guide is not included in our manual, so we will reflect it in the future. Following your excellent ROS 1 course, we look forward to your ROS 2 course. If you encounter any issues, please report them. :)

tom-howard, 2025-01-31:
@GyuH13 - Brilliant, that works thank you!

@sunghowoo - Many thanks, it would be great to get this added to the manual. And thanks for your kind feedback too. I'm still putting together the ROS2 course, but it's [available here](https://tom-howard.github.io/com2009/) if you want to take a look at where it's at so far! (ROS2 has been a steep learning curve!)

Any comments on my question regarding the robot model not appearing in RViz, or has this feature just been removed?

It's great to know that I have your support if required, thanks so much!

Tom

sunghowoo, 2025-01-31:
@tom-howard  
The robot model is not currently included in the RViz2 configuration used for running Cartographer. If you want to add it, you can do so by adding a RobotModel element and setting the Description field to /robot_description, as shown in the screenshot.

[image]

tom-howard, 2025-02-03:
Amazing, thank you!

closed 2025-02-03 by tom-howard