ROBOTIS-GIT/turtlebot3 #1052: base_footprint frame is misaligned wrt. odom frame as the start of bringup

ab31mohit, 2025-02-05:
- I'm using TURTLEBOT3 Burger hardware with ROS2-humble. I've done the setup correctly and tested everything. 
  
- But when i'm running the bringup file, then in the rviz there is some misalignment between the frames `base_footprint` and 'odom' somehow. I tried restarting and running again multiple times, even updated OpenCR firmware multiple times, still the issue is there.    
- Is it supposed to be like this? Or what should the initial transform between the 2 frames i mentioned?   
- I want somohow to align the `base_footprint` and `odom` frame or in least case wanted to have a constant initial transform (with any yaw).   
- How to resolve this? 
  

[image]

ab31mohit, 2025-02-07:
Here's an update i found. I think the issue is arising because of incorrect wheel joint position at the bringup.    
This is how the joint frames look in (simulation (where everything is correct) on bringup :   

[image]    

But in hardware when running bringup, we can see the joint frames of wheels are shifted : 

[image]    

in `odometry.cpp` file, the odometry (x,y,yaw) are being calculated wrt the wheel_l and wheel_r variables (which defined how much that specific wheel has rotated) which itself depends on diff_joint_position[0], diff_joint_position[1] and these diff_joint position variables depend on the joint_state->position[0], joint_state->position[1] variables.    
My guess is since the initial joint state position of wheels is somehow shifted, it is creating a fault odometry result at the start.   
Maybe we can force these joint positions to be zero wrt. their parent and try that.   
I'm still now 100% sure. I'm still waiting for some response from your side.

GyuH13, 2025-02-10:
Hi @ab31mohit 
We have identified the issue. There is an initial offset error between odom and base_footprint. I will address it as soon as possible and get back to you. Thank you!

GyuH13, 2025-02-14:
Hi @ab31mohit 
We have fixed the bug you mentioned. 1941f917be41a95c245761d095ab75583d2ca299
Thank you. If you have no further questions, we will close this issue soon.

ab31mohit, 2025-02-15:
Hey @GyuH13 
Thanks for replying and i'm sorry for my late reply.
I pulled the changes and after running the updated code using turtlebot3_node and turtlebot3_bringup, although the position drift is removed but there is still a lot of drift in the orientation.    
Here is the result of your updated code : 

[image]

As you can see, the robot frame (base_footprint) is still misaligned wrt. odom.  
Here is the data of /odom topic : 

[image]

Can you please try looking at this orientation drift as well.

GyuH13, 2025-02-17:
Hi @ab31mohit 
I have identified an issue with IMU drift. After reviewing the source code, I found that, similar to the odometry issue, there is a problem with initialization during bringup. When manually moving the TurtleBot3 after starting it, the odometry orientation drifts according to the movement. I will update and fix this issue. Until then, please avoid manually moving the robot after startup, and the IMU drift should not occur.
Thanks

GyuH13, 2025-02-19:
Hi @ab31mohit 
The odom offset bug has been resolved and has been merged into the main branch.
You can use it on the main branch until it is released to the humble release branch.
Now, when you turn off and on the bring-up, the odom will be reset.
Thank you for raising the issue. If you have any further questions, feel free to ask. If there are no additional questions, I will close this issue soon.

ab31mohit, 2025-02-21:
Hi @GyuH13 
Thanks for the updates. The issue is solved and the base_footprint frame is perfectly aligned at the startup now.

GyuH13, 2025-02-24:
This issue has been resolved, so I will close it. Thank you!

closed 2025-02-24 by GyuH13