ROBOTIS-GIT/turtlebot3 #1066: The frame_id /base_footprint should not start with a /

xipengwang, 2025-02-20:
On humble branch, when run `ros2 launch turtlebot3_cartographer cartographer.launch.py`, `turtlebot3_ros` node crashes due to
```
[cartographer_node-1] F0219 23:53:55.548945 10392 sensor_bridge.cpp:33] Check failed: frame_id[0] != '/' ('/' vs. '/') The frame_id /base_footprint should not start with a /. See 1.7 in http://wiki.ros.org/tf2/Migration.
[cartographer_node-1] [FATAL] [1740027235.549483888] [cartographer logger]: F0219 23:53:55.000000 10392 sensor_bridge.cpp:33] Check failed: frame_id[0] != '/' ('/' vs. '/') The frame_id /base_footprint should not start with a /. See 1.7 in http://wiki.ros.org/tf2/Migration.
```

A simple solution is just modifying `src/turtlebot3/turtlebot3_bringup/launch/robot.launch.py` to remove `/`
```
Node(
    package='turtlebot3_node',
    executable='turtlebot3_ros',
    parameters=[
        tb3_param_dir,
        {'odometry.frame_id': PythonExpression(['"', namespace, 'odom"'])},
        {'odometry.child_frame_id': PythonExpression(
            ['"', namespace, 'base_footprint"'])}],
    arguments=['-i', usb_port],
    output='screen'),
```

I didn't see other people posting a similar issue. I am wondering if I had missed anything when bring up the turtlebot3 burger?

GyuH13, 2025-02-24:
Hi @xipengwang 
A bug occurred during the process of adding the Multi Robot feature. The bug has been fixed and uploaded to the main branch. It will be included in the next release for the Humble branch, so please use the main branch until then. Thank you!

GyuH13, 2025-03-11:
Since there has been no further discussion, I will assume the issue has been resolved and proceed to close it.
Thanks

closed 2025-03-11 by GyuH13