ROBOTIS-GIT/turtlebot3 #1069: Odometry always has zero covariance

TakShimoda, 2025-03-05:
1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [x] etc (Please specify your OS here) Ubuntu 20.04.2 LTS

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - [All the commands from the quickstart guide, until the bringup, using the Noetic tab](https://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#bringup)
    - `ros2 launch turtlebot3_bringup robot.launch.py`

I'm trying to collect odometry data as well as measurements (with a camera that's separate from what's provided in the turtlebot). I collect the odometry data by subscribing to the `/odom` topic which is of type nav_msgs/msg/Odometry, with a geometry_msgs/msg/PoseWithCovariance field that has the 36 entry covariance. However, it's always zero no matter how I move the robot. I also disabled IMU to make it accumulate drift more with `ros2 param set /diff_drive_controller odometry.use_imu False`, but it still gives 0 covariances. 

As an aside, I had an issue of very large odometry values (in the magnitude of 10e204) which is the same as [here](https://github.com/ROBOTIS-GIT/turtlebot3/issues/880) and confirmed [here](https://github.com/ROBOTIS-GIT/turtlebot3/issues/926#issuecomment-1403244766). Instead of installing the new OpenCR firmware like in this latter issue, I fixed the issue that was recommended by another user by adding [these lines](https://github.com/paolorugg/my_turtlebot3_node/blob/main/tbt3_node/turtlebot3_node/src/odometry.cpp#L108-L111) and [these lines](https://github.com/paolorugg/my_turtlebot3_node/blob/main/tbt3_node/turtlebot3_node/src/odometry.cpp#L283-L288) to odometry.cpp and [this](https://github.com/paolorugg/my_turtlebot3_node/blob/main/tbt3_node/turtlebot3_node/include/turtlebot3_node/odometry.hpp#L64) and [this](https://github.com/paolorugg/my_turtlebot3_node/blob/main/tbt3_node/turtlebot3_node/include/turtlebot3_node/odometry.hpp#L73) to odometry.hpp, and then use this command `ros2 topic pub -1 /pose_relocalization geometry_msgs/Point`. Essentially this manually sets all 0's to the robot's initial position and it fixed the issue. I'm certain this shouldn't affect covariance as I don't modify anything other than the initial position with this command.

I was wondering how I could set these covariance values as I want to derive information matrices for consecutive odom poses. Could I make calculations based off the motion model for odometry? And would the off-diagonals be zeros?

Thanks

GyuH13, 2025-03-11:
Hi @TakShimoda 
The odometry initialization issue has been resolved. If you clone the Humble branch again, you should be able to see the initialized odom during bring-up. 

The TurtleBot3 only uses wheel encoder-based odometry. In odometry.cpp, the source that provides the covariance values for the odom topic is commented out. If you'd like to customize the covariance, you can modify that part of the code.
Thank you.

TakShimoda, 2025-03-12:
Thanks Gyu.

I'm assuming I can just take the constant values from [here](https://github.com/ROBOTIS-GIT/turtlebot3/blob/main/turtlebot3_node/src/odometry.cpp#L170-L183). There seems to be a to do to make it more accurate and possibly variable.

Do you have any pointers for resources I can try to implement to make it more accurate? I notice the quality in odom can vary at times, where RMSE for absolute pose error (APE) compared to ground-truth for the same movements can vary quite a lot, possibly due to slippage.

Thanks

sunghowoo, 2026-01-01:
@TakShimoda  The original question appears to be resolved. Please generate and post a follow-up question if needed.
This thread will be closed due to a long period of inactivity.