ROBOTIS-GIT/turtlebot3 #1070: map can not load in Rviz

an1ya1, 2025-03-06:
I use the waffle model,
OS：Ubuntu 22.04
ROS:humble

[image]

GyuH13, 2025-03-11:
Hi @an1ya1 
If you could describe the situation in the proper format, we can assist you more quickly.
Did you create the map using SLAM?
Which command did you use?
If an error appears in RViz saying that it failed to receive the map information, it's most likely due to a path issue.
Please provide more details about your situation so we can assist you better.

Chillhopper, 2025-03-17:
yes I have this same issue right now

GyuH13, 2025-03-17:
@Chillhopper 
If you could explain your situation in more detail, it would be easier for me to help you.

Chillhopper, 2025-03-18:
@GyuH13 

I have attached 2 screenshots, the map frame seems to be missing im not sure how to fix it. I have saved the map at home named map.pgm as per the tutorial so I dont think its a path issue. I have rosdepped a few times as well. Ive installed all the nav2 packages. Still getting error no map received

[image]

[image]

GyuH13, 2025-03-18:
Have you used the use_sim_time parameter?
Check the timestamp using `ros2 topic echo /map`

Chillhopper, 2025-03-18:
@GyuH13 

``$ export TURTLEBOT3_MODEL=burger
$ ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py

$ export TURTLEBOT3_MODEL=burger
$ ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True map:=$HOME/map.yaml
``

These are the commands I used, use_sim_time is true, for the turtlebot3_world the sim time is true on default, I am just using the commands from the manual @ https://emanual.robotis.com/docs/en/platform/turtlebot3/nav_simulation/

GyuH13, 2025-03-18:
[image]
Could you show me the pannel like above?

GyuH13, 2025-03-18:
@Chillhopper 
Try changing the parameter use_sim_time to true in the YAML file in the turtlebot3_navigation2 folder.

Chillhopper, 2025-03-18:
@GyuH13 
Is this what you mean? I changed the use_sim_time params to all true for burger, but still the same issue

[image]

Chillhopper, 2025-03-19:
@GyuH13 
Hello I have solved this issue, but I had to make a custom launch file for this. The issue lies in the map path. The steps in the Turtlebot3 tutorial don't seem to work for me, namely the `map:=$HOME/map.yaml` instruction in the tb3 navigation simulation tutorial despite me having the map.yaml file in my home directory. I think this is a recurring issue others have as well.

However in a custom launch file by specifying the path in the map, we are able to let the `bringup_launch.py` do the handling for the path. I was also previously under the impression that bringup was not needed for simulation but this seems to work for me. The bringup isnt mentioned in the tutorial as well for turtlebot3 nav2 simulation so maybe its something you can look into. 

```
map_file = os.path.join(get_package_share_directory('map_package'), 'all_maps', 'map.yaml')

IncludeLaunchDescription(
    PythonLaunchDescriptionSource([get_package_share_directory('nav2_bringup'),'/launch','/bringup_launch.py']),
    launch_arguments={
    'use_sim_time': 'true',
    'map':map_file}.items()
)
```

GyuH13, 2025-03-19:
@Chillhopper 
Glad to hear that you’ve resolved the issue.
Bringup is a process used to launch ROS nodes on the robot’s SBC. In simulation, however, Gazebo handles this part by publishing and subscribing to the necessary topics, so using bringup is not required.

If anyone else is experiencing the same issue, try changing the default value of the map parameter in the navigation.launch file within the turtlebot3_navigation2 package to the correct path of your own map file.

GyuH13, 2025-05-20:
This issue is considered resolved and will be closed.

closed 2025-05-20 by GyuH13