ROBOTIS-GIT/turtlebot3 #1072: Cant run the package, not sure why. I followed the manual

Chillhopper, 2025-03-12:
`kaushik@msi:~/turtlebot3_ws$ export TURTLEBOT3_MODEL=burger
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
[INFO] [launch]: All log files can be found below /home/kaushik/.ros/log/2025-03-12-22-30-59-678487-msi-11839
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
urdf_file_name : turtlebot3_burger.urdf
[INFO] [gzserver-1]: process started with pid [11840]
[INFO] [gzclient-2]: process started with pid [11842]
[INFO] [robot_state_publisher-3]: process started with pid [11844]
[INFO] [spawn_entity.py-4]: process started with pid [11846]
[robot_state_publisher-3] [INFO] [1741789860.275825472] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1741789860.275907280] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1741789860.275916189] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-3] [INFO] [1741789860.275921307] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-3] [INFO] [1741789860.275925893] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1741789860.275930451] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-3] [INFO] [1741789860.275934648] [robot_state_publisher]: got segment wheel_right_link
[spawn_entity.py-4] [INFO] [1741789860.561582416] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1741789860.561773715] [spawn_entity]: Loading entity XML from file /home/kaushik/turtlebot3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf
[spawn_entity.py-4] [INFO] [1741789860.562269610] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1741789860.562423179] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [ERROR] [1741789890.623214225] [spawn_entity]: Service /spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[spawn_entity.py-4] [ERROR] [1741789890.623403245] [spawn_entity]: Spawn service failed. Exiting.
[ERROR] [spawn_entity.py-4]: process has died [pid 11846, exit code 1, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity burger -file /home/kaushik/turtlebot3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf -x -2.0 -y -0.5 -z 0.01 --ros-args'].`

jhill515, 2025-03-16:
I had a similar issue centered around launching turtlebot3_gazebo. Make sure you re-`source` your `install/setup.bash` file after building and ensure every terminal defines `TURTLEBOT3_MODEL`, `LDS_MODEL`, and `ROS_DOMAIN_ID`.

GyuH13, 2025-03-17:
Hello @Chillhopper 
I recently experienced a similar error.
Is it a situation where the map and Gazebo are working fine, but only the robot model is not being spawned?
If spawn_entity.py is not being executed, check whether ROS 2 communication is working properly.
In my case, the issue occurred because ROS 2 communication temporarily didn't work well, so the spawn service wasn't being published to Gazebo.
After I changed the ROS_DOMAIN_ID number, it started working normally.
Try it once and let me know how it goes!

Chillhopper, 2025-03-17:
Hello @GyuH13 After turning my PC off and on it seems to work, not sure why. I just had to keep running the script several times before it could work

GyuH13, 2025-03-17:
@Chillhopper
`Spawn_entity.py` publish a service for spawning entity in Gazebo, but sometimes ROS 2 communication doesn't work properly. Most likely, when you encountered the issue, you were not able to see any topics using the`ros2 topic list` command. Glad to hear it's been resolved. I’ll go ahead and close this issue.

closed 2025-03-17 by GyuH13