ROBOTIS-GIT/turtlebot3 #1073: Bringup throws exception at launch

jhill515, 2025-03-16:
# Issue
`turtlebot3_bringup` throws `rclcpp:exceptions::RCLError` exception at launch

# Steps to Recreate
**Note:** This was all executed through the `osrf/ros:humble-desktop-full` container running on a Ubuntu 22.04 workstation. Launch of the image used: `docker run -it -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix --gpus all osrf/ros:humble-desktop-full`. All *attached* Docker instances used: `docker exec -it [CONTAINER_ID] bash`

1. Launch ROS Humble Desktop image. Follow PC Setup & SBC Setup starting from Section 3.3-3.3.1 (Note: without LDS configuration, Gazebo simulation crashes immediately).
2. Set `TURTLEBOT3_MODEL=burger`, `LDS_MODEL=LDS-02`, and `ROS_DOMAIN_ID=30`.
3. Launch Gazebo simulation: `ros2 launch turtlebot3_gazebo turtlebot3_house.launch.py`. Allow application to load (i.e., wait a couple minutes)
4. Open another terminal and attach to running Docker instance. Source `~/turtlebot3_ws/setup.bash`. Set environment variables (see Step 2).
5. Launch Bringup: `ros2 launch turtlebot3_bringup robot.launch.py`

# Observed Result
```
root@4c4065d0602b:~/turtlebot3_ws# ros2 launch turtlebot3_bringup robot.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-03-16-21-02-20-806084-4c4065d0602b-20019
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
[INFO] [robot_state_publisher-1]: process started with pid [20020]
[INFO] [ld08_driver-2]: process started with pid [20022]
[INFO] [turtlebot3_ros-3]: process started with pid [20024]
[turtlebot3_ros-3] [INFO] [1742158940.880334040] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [1742158940.880471349] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [ERROR] [1742158940.880522706] [DynamixelSDKWrapper]: Failed to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [ERROR] [1742158940.880526833] [DynamixelSDKWrapper]: Failed to initialize SDK handlers
[turtlebot3_ros-3] [ERROR] [1742158940.880537874] [turtlebot3_node]: Failed connection with Devices
[turtlebot3_ros-3] [PortHandlerLinux::SetupPort] Error opening serial port!
[turtlebot3_ros-3] [INFO] [1742158940.880835905] [turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [1742158940.880860281] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [1742158940.880873486] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[turtlebot3_ros-3]   what():  could not create publisher: rcl node's context is invalid, at ./src/rcl/node.c:428
[ld08_driver-2] Can't find LDS-02
[robot_state_publisher-1] [INFO] [1742158940.901238254] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1742158940.901297024] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1742158940.901306522] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-1] [INFO] [1742158940.901312654] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-1] [INFO] [1742158940.901318415] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-1] [INFO] [1742158940.901323604] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-1] [INFO] [1742158940.901327361] [robot_state_publisher]: got segment wheel_right_link
[ERROR] [turtlebot3_ros-3]: process has died [pid 20024, exit code -6, cmd '/root/turtlebot3_ws/install/turtlebot3_node/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 --ros-args -r __ns:=/ --params-file /root/turtlebot3_ws/install/turtlebot3_bringup/share/turtlebot3_bringup/param/burger.yaml --params-file /tmp/launch_params_1ogl6qb3 --params-file /tmp/launch_params_gpad8p3b'].
[INFO] [ld08_driver-2]: process has finished cleanly [pid 20022]
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-1] [INFO] [1742158949.165326700] [rclcpp]: signal_handler(signum=2)
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 20020]
```

# Expected Result
Continuous execution without thrown exception

# Other Notes
I am creating a simulation environment for students to use TurtleBot3 Burger in an indoor environment. Because of laboratory rules, all of this must be done through some Docker container (yay Security Theater!). If I'm skipping around too much or going about this the wrong way, your pointing direction is extremely helpful!

GyuH13, 2025-03-17:
Hello @jhill515 
The bringup package is used for the actual robot and is not required when running in the Gazebo simulation environment! The bringup package depends on the LiDAR package. If you are using a simulation environment, you can simply run the simulation launch file without needing the bringup package.

jhill515, 2025-03-17:
Hi @GyuH13 ,

Thank you for your guidance. For clarification, I intend for the group to be able to execute SLAM features on the simulated machine. So, I was unclear if I needed to run the Bring Up. Does that change anything?

jhill515, 2025-03-17:
@GyuH13 , please disregard my last question, I figured it out for myself.

To the community, I don't want to leave anyone hanging. I just followed [3.1 PC Setup](https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#pc-setup) and [6.1 Gazebo Simulation](https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#gazebo-simulation). No other instructions need to execute. Do beware that you are setting `ROS_DOMAIN_ID` and `TURTLEBOT3_MODEL` and additionally `source /usr/share/gazebo/setup.sh` in every bash instance of the container.

closed 2025-03-17 by jhill515

GyuH13, 2025-03-18:
@jhill515 
Bring Up is for running the actual robot. In simulation, you can operate the robot simply by starting the simulation!
If you want to run SLAM in simulation, just launch the simulation and then launch the SLAM process.