ROBOTIS-GIT/turtlebot3 #1076: TB3

kimudonghyeon, 2025-03-21:
I am trying to use TurtleBot3 for SLAM, but instead of controlling it via the keyboard, I want the robot to autonomously drive and create a map using SLAM.

The code I have written for autonomous driving is as follows:
----------------------------------------------------------------------------------------------------------------
import rclpy
from rclpy.node import Node
from geometry_msgs.msg import PoseStamped
from nav2_simple_commander.robot_navigator import BasicNavigator
import random
import time

class AutoExplore(Node):
    def __init__(self):
        super().__init__('auto_explore')
        self.navigator = BasicNavigator()
        self.exploration_running = True  # Exploration state

    def explore(self):
        self.get_logger().info('Exploration started')

        # Automatically generates goals and moves the robot
        while self.exploration_running:
            self.get_logger().info('Searching for a new area to explore')
            goal = self.get_random_pose()  # Set the goal to an unexplored area
            self.navigate_to_goal(goal)
            time.sleep(5)  # Wait long enough for the robot to move

        self.get_logger().info('Exploration completed!')

    def get_random_pose(self):
        """
        Generates a random goal position in an unexplored area
        """
        pose_stamped = PoseStamped()
        pose_stamped.header.frame_id = "map"  # Set the coordinate frame
        pose_stamped.header.stamp = self.get_clock().now().to_msg()  # Set the current time
        pose_stamped.pose.position.x = random.uniform(-5, 5)  # Generate a random x coordinate
        pose_stamped.pose.position.y = random.uniform(-5, 5)  # Generate a random y coordinate
        pose_stamped.pose.orientation.w = 1.0  # Set orientation to default
        return pose_stamped

    def navigate_to_goal(self, goal):
        """
        Moves the robot to the goal position
        """
        self.get_logger().info(f'Moving to goal: {goal.pose.position.x}, {goal.pose.position.y}')
        self.navigator.goToPose(goal)  # Use 'PoseStamped' message
        
        # Check if the robot reached the goal (using wait_for_result)
        result = self.navigator.wait_for_result()  # Check status
        if result:
            self.get_logger().info(f'Successfully reached goal: {goal.pose.position.x}, {goal.pose.position.y}')
        else:
            self.get_logger().warning(f'Failed to reach goal: {goal.pose.position.x}, {goal.pose.position.y}')

def main(args=None):
    rclpy.init(args=args)
    auto_explore = AutoExplore()

    # Start the automatic exploration
    auto_explore.explore()

    # Shutdown ROS2 when done
    rclpy.shutdown()

if __name__ == '__main__':
    main()
----------------------------------------------------------------------------------------------------------------

However, when I run the code, the robot does not move autonomously.
[... 38 lines cut ...]
[component_container_isolated-1] [INFO] [1742542932.757873799] [controller_server]: Destroying bond (controller_server) to lifecycle manager.
[component_container_isolated-1] [INFO] [1742542932.757882805] [lifecycle_manager_localization]: Running Nav2 LifecycleManager rcl preshutdown (lifecycle_manager_localization)
[component_container_isolated-1] [INFO] [1742542932.757897369] [lifecycle_manager_localization]: Terminating bond timer...
[component_container_isolated-1] [INFO] [1742542932.780918898] [amcl]: Message Filter dropping message: frame 'base_scan' at time 1742542931.776 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742542932.790421658] [amcl]: Running Nav2 LifecycleNode rcl preshutdown (amcl)
[component_container_isolated-1] [INFO] [1742542932.790500785] [amcl]: Deactivating
[component_container_isolated-1] [INFO] [1742542932.790520440] [amcl]: Destroying bond (amcl) to lifecycle manager.
[component_container_isolated-1] [INFO] [1742542932.800727403] [amcl]: Cleaning up
[component_container_isolated-1] [INFO] [1742542932.822197104] [amcl]: Destroying bond (amcl) to lifecycle manager.
[component_container_isolated-1] [INFO] [1742542932.822232628] [map_server]: Running Nav2 LifecycleNode rcl preshutdown (map_server)
[component_container_isolated-1] [INFO] [1742542932.822253437] [map_server]: Deactivating
[component_container_isolated-1] [INFO] [1742542932.822268347] [map_server]: Destroying bond (map_server) to lifecycle manager.
[component_container_isolated-1] [INFO] [1742542932.832387953] [map_server]: Cleaning up
[component_container_isolated-1] [INFO] [1742542932.832479561] [map_server]: Destroying bond (map_server) to lifecycle manager.
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT) again, ignoring...
[component_container_isolated-1] [INFO] [1742542934.008697726] [rclcpp]: signal_handler(signum=2)
[ERROR] [component_container_isolated-1]: process[component_container_isolated-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [component_container_isolated-1]: sending signal 'SIGTERM' to process[component_container_isolated-1]
[component_container_isolated-1] [INFO] [1742542936.463564371] [rclcpp]: signal_handler(signum=15)
[ERROR] [component_container_isolated-1]: process[component_container_isolated-1] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [component_container_isolated-1]: sending signal 'SIGKILL' to process[component_container_isolated-1]
[ERROR] [component_container_isolated-1]: process has died [pid 14231, exit code -9, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_isolated --ros-args --log-level info --ros-args -r __node:=nav2_container --params-file /tmp/launch_params_kt2tir8z --params-file /tmp/launch_params_zte2cqhw -r /tf:=tf -r /tf_static:=tf_static'].
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Thank you.

chan-1207, 2025-04-09:
I've checked with similar code in the environment you describe and there are no issues. I can't help you with your individual code, but it looks like there is an issue with the code you wrote. Please try writing your code using the Nav2 API as a reference.

https://docs.nav2.org/

GyuH13, 2025-04-30:
This issue is considered resolved and will be closed.
Thanks

closed 2025-04-30 by GyuH13