ROBOTIS-GIT/turtlebot3 #1079: TurtleBot3 Compatibility Issue with ROS 2 Jazzy (Ubuntu 24.04)

Armstrongdematic, 2025-03-26:
I am trying to run TurtleBot3 on ROS 2 Jazzy (Ubuntu 24.04), but it is not officially supported. I cloned the turtlebot3 and turtlebot3_msgs repositories from GitHub and built them manually.

shiba-8ro, 2025-03-26:
@Armstrongdematic 
Is the new Jazzy branch useful to you?
https://github.com/ROBOTIS-GIT/turtlebot3/tree/feature-jazzy

Armstrongdematic, 2025-03-26:
Thank you for sharing the feature-jazzy branch! I will test it on ROS 2 Jazzy (Ubuntu 24.04) and provide feedback.
Is there anything specific I should check or any known issues in this branch?

shiba-8ro, 2025-03-26:
The information you need is in below link.
https://github.com/ROBOTIS-GIT/turtlebot3/pull/1077

Armstrongdematic, 2025-03-28:
I cloned Jazzy Jalisco TurtleBot3 package and tried running it in simulation, but I’m facing some issues. The bot is visible, but it’s not responding when I use turtlebot3_teleop to move it. Additionally, I’m getting errors where the hlds_laser_publisher and turtlebot3_ros processes crash with exit codes 255 and -6. I'm not using any hardware; I just want to run everything in simulation, including navigation. Could you help me resolve these issues? Also, if possible, could you share a video demonstrating a working setup? 

Thanks in advance!"

GyuH13, 2025-03-31:
Hello, @Armstrongdematic 
The Jazzy version is currently being prepared for support.
There are some changes, and we will update it along with the manual as soon as possible. Additionally, starting from Jazzy, we plan to support GZ Sim. The details will be documented in the e-manual. Thank you.

Armstrongdematic, 2025-03-31:
Hi @GyuH13 ,

Thank you for the update! I appreciate the information about the Jazzy version and the upcoming support for GZ Sim. I look forward to seeing the new manual and any additional details.

brean, 2025-03-31:
Hey guys,
@GyuH13 thanks a lot for the updates, so far I've build my own debian packages and used them on the turtlebot 3 without any jazzy-specific modifications, for what I tested the current main or humble-branches worked fine as stand-ins ( https://github.com/brean/turtlebot3-docker-build ).
I am also working with turtlebot (2/3) and gz-sim, however I am just using the default ros2 differential drive example and add the turtlebot3-urdf from turtlebot3_description into that, an officially supported turtlebot 3 gz-sim would be nice to see and is probably not much work as the original turtlebot3-gazebo-classic was also quite close to the diff-drive example.
If you need any support or need to do some specific testing or other support I would be happy to help.

Armstrongdematic, 2025-04-01:
Hey @brean,

Thanks for sharing your experience and the Debian packages! That’s really helpful. 

I’m currently working with TurtleBot3 on ROS 2 Jazzy, and I ran into some issues getting it to move in RViz without Gazebo. I saw that you used the Humble branches without any Jazzy-specific modifications—did you have to tweak anything in the launch files or parameter configurations to make it work?

Also, your approach with GZ-SIM and the differential drive example sounds interesting. Could you share how you integrated the TurtleBot3 URDF into GZ-SIM? I’d love to test it out.

Looking forward to your insights!

brean, 2025-04-01:
I did not tweak anything, just build the .deb-packages from the checked-out humble/main branches in the versions from the git submodule you see on my repo. I tested getting the laserscan and controlling using a playstation dualshock 4 controller, both worked nicely. I did not test autonomous navigation, however my students wanted to check that, but I will update the packages first. It even worked with running rviz on humble and connecting to the robot (you get some warnings in the terminal but it still receives data from the lidar).
For gz-sim I published this for the turtlebot 2/kobuki base I can recommend this as entry-point for a turtlebot 3 environment: https://github.com/helloric/kobuki_gzsim
In teaching we use our real robots and don't use gazebo right now so I only have the older simulated version we used during the Corona-Lockdowns based on ROS 2 foxy, however if I remember correctly it did not work with the current version of gz-sim as I had some issues with libignition, but those should be solved in the current ros2-jazzy. 
https://github.com/brean/ros2-turtlebot3-gazebo-docker/tree/foxy-gz-sim-garden

brean, 2025-04-01:
I've just build the packages from the current jazzy-branches and tried to start everything on the robot, however now I am getting the error:
```
[DynamixelSDKWrapper]: Failed to read[[TxRxResult] Port is in use!]
```
So I can not control the robot anymore.
However the port (/dev/ttyACM0) is not used by anything else as far as I can see and the user should have the rights to access it as the user is in the dialout-group just as the /tty/ACM0.

Armstrongdematic, 2025-04-02:
Hello @brean,

Thanks for the details! It’s great to hear that you got everything working with the Humble branches. I’ll try building the .deb packages the same way and see if I can get TurtleBot3 moving in RViz.

For GZ-SIM, I’ll check out the kobuki_gzsim repo you shared. It sounds like a good starting point. Did you run into any compatibility issues when integrating TurtleBot3’s URDF into the latest gz-sim versions? I’d love to test it out on Jazzy.

Also, regarding the DynamixelSDKWrapper error you’re seeing—I've run into similar issues with /dev/ttyACM0 before. Have you tried checking if another process might be holding onto the port using,

**sudo lsof /dev/ttyACM0**

If anything shows up, you might need to kill that process.

brean, 2025-04-02:
After updating the firmware and installing `ros-jazzy-dynamixel-sdk` (which is not a dependency on the ros-packages) I am not getting any errors anymore but the robot is still not moving if I send something on `/cmd_vel`. Any other idea how to 
```bash
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
[INFO] [robot_state_publisher-1]: process started with pid [5116]
[INFO] [hlds_laser_publisher-2]: process started with pid [5117]
[INFO] [turtlebot3_ros-3]: process started with pid [5118]
[hlds_laser_publisher-2] [INFO] [1743586961.642952729] [hlds_laser_publisher]: Init hlds_laser_publisher Node Main
[hlds_laser_publisher-2] [INFO] [1743586961.647541190] [hlds_laser_publisher]: port : /dev/ttyUSB0 frame_id : base_scan
[turtlebot3_ros-3] [INFO] [1743586961.951294960] [turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [1743586961.954556085] [turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [INFO] [1743586961.960229280] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [INFO] [1743586961.965760184] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[turtlebot3_ros-3] [INFO] [1743586962.002587615] [turtlebot3_node]: Start Calibration of Gyro
[robot_state_publisher-1] [INFO] [1743586962.019318242] [robot_state_publisher]: Robot initialized
[turtlebot3_ros-3] [INFO] [1743586967.003235605] [turtlebot3_node]: Calibration End
[turtlebot3_ros-3] [INFO] [1743586967.003543937] [turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [1743586967.005515487] [turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [1743586967.007466934] [turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] [INFO] [1743586967.074945127] [turtlebot3_node]: Succeeded to create battery state publisher
[turtlebot3_ros-3] [INFO] [1743586967.102796888] [turtlebot3_node]: Succeeded to create imu publisher
[turtlebot3_ros-3] [INFO] [1743586967.139319119] [turtlebot3_node]: Succeeded to create sensor state publisher
[turtlebot3_ros-3] [INFO] [1743586967.155677407] [turtlebot3_node]: Succeeded to create joint state publisher
[turtlebot3_ros-3] [INFO] [1743586967.156117561] [turtlebot3_node]: Add Devices
[turtlebot3_ros-3] [INFO] [1743586967.156287404] [turtlebot3_node]: Succeeded to create motor power server
[turtlebot3_ros-3] [INFO] [1743586967.176004995] [turtlebot3_node]: Succeeded to create reset server
[turtlebot3_ros-3] [INFO] [1743586967.188324453] [turtlebot3_node]: Succeeded to create sound server
[turtlebot3_ros-3] [INFO] [1743586967.201281772] [turtlebot3_node]: Run!
[turtlebot3_ros-3] [INFO] [1743586967.474167962] [diff_drive_controller]: Init Odometry
[turtlebot3_ros-3] [INFO] [1743586967.570837856] [diff_drive_controller]: Run!
```
So I think it looks good, I am just using this command to send something (on the turtlebot itself):
```
ros2 topic pub /cmd_vel geometry_msgs/Twist "{linear: {x: 1.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.0}}" -r 10
```
Any ideas how to debug this further?

GyuH13, 2025-04-02:
@brean 
Hello
Since the official update of the Jazzy version, the cmd_vel topic now uses the TwistStamped type.
If you have updated your Jazzy TB3, that is likely the reason for this change.
Check
```
ros2 topic info /cmd_vel
```

brean, 2025-04-02:
Thanks @GyuH13!

Sending
```ros2 topic pub /cmd_vel geometry_msgs/TwistStamped "twist: {linear: {x: 1.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.0}}"```
 worked!
I am aware that this sends `stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id=''` which is not ideal, but the robot is moving.

GyuH13, 2025-04-02:
@brean 
In Nav2, the cmd_vel topic has been changed to use the TwistStamped type. Accordingly, from the Jazzy version, TB3 has also set TwistStamped as the default.

If you want to use the previous Twist type, you can clone the package source and set the `enable_stamped_cmd_vel` parameter to false in the turtlebot3_bringup package.
Thanks

brean, 2025-04-03:
Quick feedback:
The turtlebot3_cartographer and turtlebot3_navigation2 are working fine. 
We were able to quickly map and autonomously navigate in our lab.
Great work everyone 🎉

[image]

brean, 2025-04-03:
So looking into gz-sim integration would probably be the next thing to do so we can close this issue.
@Armstrongdematic did you made any progress or should I dust-off my foxy code and try to build something?

Armstrongdematic, 2025-04-04:
Hello Everyone!,

@brean, great to hear that turtlebot3_cartographer and turtlebot3_navigation2 are working well! As for gz-sim integration, I’ve been working on it in ROS 2 Jazzy and have the basic simulation up and running. The TurtleBot3 is successfully spawning and moving inside Gazebo, but I’m still tweaking things like TF alignment and navigation stability.

For waypoint following, I need to run,

**ros2 launch nav2_bringup navigation_launch.py**

Also, I’m using Jazzy exclusively, so I won’t be working with Foxy. I’ll keep updating as I make more progress!

I’ve set things up using the turtlebot3_simulations repository (which I cloned) instead of launching **gz_sim** directly. Still working through some refinements, but things are coming together well!

Looking forward to seeing how this all comes together.

GyuH13, 2025-04-07:
@Armstrongdematic Hello.
You can use the jazzy branch in the TurtleBot repository to run Gazebo simulation and Nav2 on Jazzy. 
The simulation-related e-manual will be uploaded soon. Thank you!

Armstrongdematic, 2025-04-07:
Hey @GyuH13,  thanks for the info!

I’m also working on Jazzy from the beginning, so this helps a lot. I’ll use the jazzy branch for turtlebot3_simulations and test with Nav2 and Gazebo.

Looking forward to the simulation e-manual. Appreciate your efforts!

brean, 2025-04-07:
I just saw that there's a "tb3"-package package in the official nav2_minimal-navigation2 repo, maybe this can be merged back into turtlebot3 (I also created an issue there to remove gz-sim from nav2 as I don't need that on my hardware)

[the nav2-minimal-tb3-sim](https://github.com/ros-navigation/nav2_minimal_turtlebot_simulation/tree/main/nav2_minimal_tb3_sim)

The issue: https://github.com/ros-navigation/navigation2/issues/5063

Armstrongdematic, 2025-04-08:
Hey @brean,

Thanks for sharing that info. I’ll check out the tb3-package in the nav2_minimal-navigation2 repo. And yeah, removing gz-sim sounds like a good move if it's not needed.

brean, 2025-04-09:
Hey @Armstrongdematic,
looks like @GyuH13 beat you to it:
https://www.youtube.com/watch?v=oqT6umwqLk8
https://github.com/ROBOTIS-GIT/turtlebot3_simulations/tree/jazzy

So I think we can close this issue?!

Armstrongdematic, 2025-04-09:
Hey @brean,
Thanks for sharing the update!,
I’ll try this out first when I get to it, and once I’ve verified everything works, I’ll go ahead and close the issue.

GyuH13, 2025-04-10:
@brean , @Armstrongdematic 
It has been a pleasure talking with all of you.
If you experience any inconvenience or find any bugs while using the Jazzy version, please feel free to let us know right away.
We will continue to do our best to provide a better TurtleBot experience.
Thank you all for your contributions!

Please close this issue if there are no further problems after using the Jazzy version.

Armstrongdematic, 2025-04-10:
Hello @GyuH13,

Thanks a lot! Really appreciate all the help and the awesome work you guys are doing with TurtleBot3 
I’ll give the Jazzy version a try soon, and if everything works fine, I’ll go ahead and close the issue.
Been great chatting with all of you.
 
Thanks again!

closed 2025-04-10 by Armstrongdematic