ROBOTIS-GIT/turtlebot3 #1082: Applying Namespace to URDF Files

Ji-Sung05, 2025-03-28:
I followed the "10. Examples > Load Multiple TurtleBot3s" tutorial from the TurtleBot3 ROBOTIS e-Manual and successfully launched two robots. The nodes and topics are properly separated by namespaces.

However, when I launched the SLAM package, I noticed that the robot models were not being rendered. I suspect this issue is caused by the fact that the frames used in the URDF files are not distinguished by namespaces.

How can I apply namespaces to the URDF files so that the robot models are properly rendered when using multiple robots?

Ji-Sung05, 2025-03-28:
I have successfully resolved the issue.

GyuH13, 2025-03-30:
Hello, @Ji-Sung05 
I'm glad to hear that you have resolved the issue.
Multi-robot SLAM is not supported yet. We will make efforts to support multi-robot SLAM and Nav2 in the future.

GyuH13, 2025-04-30:
This issue has been resolved and is now closed.
If you encounter the same error, please make sure to set the PREFIX under the ROBOT MODEL panel in RViz. This will allow the model to display correctly in RViz.
Thanks

closed 2025-04-30 by GyuH13