ROBOTIS-GIT/turtlebot3 #1085: How can I increase the speed limit of turtlebot3 on simulation? The default value is 0.22 to 0.26m/s

baiyeha, 2025-04-04:
When turtlebot3 waffle_pi is used for navigation in simulation, the upper limit of linear x is set at 1.0m/s, but the actual linear x can only reach about 0.5m/s. Is this the limitation of turtlebot3 simulation model itself? For example, DynamixelSDK Settings or the wheel speed limit of the simulation turtlebot3, how can I increase the speed limit of the simulation turtlebot3 waffle_pi?

shiba-8ro, 2025-04-04:
For reference, Physical Turtlebot3 Waffle Pi max speed is 0.26 m/s. Please provide your ROS version for better support from ROBOTIS engineers.

shiba-8ro, 2025-04-04:
Have you tried changing the velocity smoother parameters?

[image]

baiyeha, 2025-04-04:
I am very sorry that I did not find the relevant configuration file you provided in the relevant package of the galactic version turtlebot3 project. I did not find this file before, so I did not modify it. This may be the problem, can you provide the location of the configuration file? Thank you very much!

shiba-8ro, 2025-04-04:
I understand that galactic-devel is currently showing a 404 error, and unfortunately, I'm unable to access its contents myself to verify what was there. As I'm primarily a TB3 user, my knowledge in this specific area is limited, and I sincerely apologize that I can't be of further assistance on this particular issue.
https://github.com/ROBOTIS-GIT/turtlebot3/tree/galactic-devel

GyuH13, 2025-04-07:
@baiyeha , Hello!
Have you resolved the issue?
As mentioned above, you can configure it through the Nav2 parameter file.

In the DWBController, set `max_vel_x` and `max_vel_xy`, and then configure `max_velocity` in the `velocity_smoother`.

For versions prior to Jazzy, `velocity_smoother` parameters may not exist in yaml file so default values will be used.
In this case, please add the following section to your parameter YAML file:
```
velocity_smoother:
  ros__parameters:
    smoothing_frequency: 20.0
    scale_velocities: false
    feedback: "OPEN_LOOP"
    max_velocity: [1.0, 0.0, 2.5]
    min_velocity: [-1.0, 0.0, -2.5]
    deadband_velocity: [0.0, 0.0, 0.0]
    velocity_timeout: 1.0
    max_accel: [2.5, 0.0, 3.2]
    max_decel: [-2.5, 0.0, -3.2]
    odom_topic: "odom"
    odom_duration: 0.1
    use_realtime_priority: false
    enable_stamped_cmd_vel: true
```

GyuH13, 2025-04-09:
Frist, check your `cmd_vel` value by below command when Robot run as max_velocity.
```
ros2 topic echo /cmd_vel
```
If `cmd_vel` value is lower than you set, check the publisher of cmd_vel by rqt-graph.
The parameters of the node that publishes cmd_vel last are likely the culprit.

GyuH13, 2025-04-29:
Since there are no further discussions, I am closing this issue. Please open a new one if you have any additional questions.
Thanks

closed 2025-04-29 by GyuH13