ROBOTIS-GIT/turtlebot3 #1086: Issues Running TurtleBot3 Simulation on ROS 2 Humble – Gazebo & RViz Failures

ApurvaNikhare, 2025-04-05:
Hi, I'm facing multiple issues while running the TurtleBot3 simulation on ROS 2 Humble. Gazebo fails to launch properly, and after cloning the TurtleBot3 repo, RViz doesn't load the robot or map correctly. I'm also unable to send navigation goals.

Environment

ROS 2 Distro: Humble Hawksbill

Ubuntu Version: 22.04

TurtleBot3 Model: burger

Installation Method: Initially via apt, then cloned from source

Gazebo Version: Installed with ROS 2 Humble

RViz Version: Installed with ROS 2 Humble

Jagadeesh-pradhani, 2025-04-06:
Hey, could you send terminal error log, so that i can understand it clearly

GyuH13, 2025-04-07:
Hello, @ApurvaNikhare 
If you could explain the situation in more detail, it would help us provide a quicker solution.
Please share the steps you took to run it, the commands you used, and any terminal errors you encountered!

ApurvaNikhare, 2025-04-10:
Hi, I'm facing some issues while running the TurtleBot3 simulation on ROS 2 Humble.

    When I try to launch Gazebo, it doesn't open properly.

    After cloning the TurtleBot3 repository and building the workspace, RViz doesn't show the robot model or map.

    I'm also not able to send navigation goals.

I tried following the instructions, but it seems something's not working as expected. I'm not sure if it's a path issue, launch config, or something else.

Let me know what exact logs or files I should share to help debug this. 

Thanks!

GyuH13, 2025-04-11:
@ApurvaNikhare 
Hello
Are you saying that Gazebo is not launching?
Could you show me your terminal error ?

ApurvaNikhare, 2025-05-06:
Hi everyone,

Thank you for your responses and willingness to help! I wanted to let you know that I was able to resolve the issues I was facing with the TurtleBot3 simulation on ROS 2 Humble.

closed 2025-05-06 by ApurvaNikhare