ROBOTIS-GIT/turtlebot3 #1087: Extend TurtleBot3 ROS 2 fw to publish pins A0-A5

travis-mendoza, 2025-04-08:
### Operating System:

Ubuntu 22.04

### ROS version :

Humble

### Turtlebot3 Model:

Burger

### Description

I want to publish the OpenCR's analog inputs A0-A5 as a ROS2 topic while running the navigation2 package on my TurtleBot3 Burger. 

### Steps to reproduce issue

1. Search for OpenCR TurtleBot3 .ino file compatible with ROS 2.
2. Get confused.
3. Create a GitHub issue to ask for help.
4. ???
5. Profit.

### What Have You Tried?

```shell
1. I checked if the OpenCR analog inputs were already published as ROS2 topics but I did not see them after I ran the turtlebot3_bringup launch file.

2. I looked for the OpenCR turtlebot3 source code and found the example 

['turtlebot3_core.ino'](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino) 

which is very clear and detailed but incompatible with ROS 2. There is a 

['turtlebot3_burger.ino'](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/examples/turtlebot3_burger/turtlebot3_burger.ino) 

in the turtlebot3_ros2 directory of this repo. I assume this is what is compiled into the 

[binary file](https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/latest/opencr_update.tar.bz2)

I installed while following the TurtleBot3 quickstart guide. However, 'turtlebot3_burger.ino' abstracts all its functions to the TurtleBot3Core. TurtleBot3Core looks like it is further abstracted in 

['turtlebot3.h'](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/include/turtlebot3/turtlebot3.h).

So how do all these files talk to each other and where should I add code to extend the firmware to read from pins A0-A5?
```

### Relevant logs / Terminal output

```shell

```

GyuH13, 2025-04-09:
Hello @travis-mendoza 
Clone `OPENCR` repo and modified here. [turtlebot3.cpp](https://github.com/ROBOTIS-GIT/OpenCR/blob/68ec75d8a400949580ecf263e0105ea9743b878e/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/src/turtlebot3/turtlebot3.cpp)
And then build and upload [turtlebot3_burger.ino](https://github.com/ROBOTIS-GIT/OpenCR/blob/68ec75d8a400949580ecf263e0105ea9743b878e/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/examples/turtlebot3_burger/turtlebot3_burger.ino)  this file into OpenCR

travis-mendoza, 2025-04-14:
Thank you! I extended [turtlebot3.cpp](https://github.com/ROBOTIS-GIT/OpenCR/blob/68ec75d8a400949580ecf263e0105ea9743b878e/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/src/turtlebot3/turtlebot3.cpp) to read from the analog pins by adding them to the control table and creating an update function for them.

I'd like to contribute back to this project if you're interested! If so, what's the best way to contribute these additions? I'd be happy to create a pull request.

I also extended the turtlebot3_node from the [turtlebot3 repo](https://github.com/ROBOTIS-GIT/turtlebot3) to publish the analog pin values to a ROS 2 topic. It's a more substantial extension because I had to create new files for it, but I hope I can contribute that to the turtlebot project, too.

GyuH13, 2025-04-15:
Hello, @travis-mendoza  
I'm glad to hear that it has been resolved.
And thank you for your contribution and passion for TurtleBot3.
I have question for your development.
Is the code you developed the one that publishes the `raw data values` coming in through A0 to A5 as a topic?
Please let me know the detailed functions! I will apply it after discussion.

travis-mendoza, 2025-04-15:
Yes, I modified turtlebot3.cpp from the [OpenCR repo](https://github.com/ROBOTIS-GIT/OpenCR) to read the raw 10-bit ADC values from the OpenCR's analog input pins A0 through A5. 

Then I made additions to the turtlebot3_node of the [turtlebot3 repo](https://github.com/ROBOTIS-GIT/turtlebot3) to publish those values as a ROS 2 topic named /analog_pins using a std_msgs/UInt16MultiArray message.

You can see the implementation of the opencr code and the turtlebot3_node in my public [turtlebot repo](https://github.com/travis-mendoza/turtlebot_repo/tree/main/feature_read_analog_pins). Right now my changes are organized in this repo for the project I am working on. But I can fork the official repos and add my changes to new branches in those forks - I assume that would simplify your review process.

travis-mendoza, 2025-04-17:
Hi @GyuH13.

I am following up from my last message. I created pull requests to enable this change:

- [OpenCR PR#342](https://github.com/ROBOTIS-GIT/OpenCR/pull/342)
- [turtlebot3 PR#1095](https://github.com/ROBOTIS-GIT/turtlebot3/pull/1095)

I hope these make your review easier.

Also, for anyone interested, I deleted the repo I mentioned in my previous comment and moved the project to [ez-turtlebot](https://github.com/travis-mendoza/ez-turtlebot)

I am closing this issue since my problem is resolved.

closed 2025-04-17 by travis-mendoza