ROBOTIS-GIT/turtlebot3 #1088: turtlebot3_navigation2 should not depend on nav2_bringup

brean, 2025-04-08:
### Operating System:

Ubuntu

### ROS version:

humble, jazzy, rolling

### Turtlebot3 Model:

burger

### Bug Description

As discussed here https://github.com/ros-navigation/navigation2/issues/5063

[turtlebot3_navigation2](https://github.com/ROBOTIS-GIT/turtlebot3/tree/main/turtlebot3_navigation2) should not rely on nav2_bringup, as it has a dependency on gazebo which we don't want to install on actual hardware.

### Steps to Reproduce

try to uninstall all gz-sim-packages, it will try to uninstall nav2 as well:
```
sudo apt remove ros-jazzy-ros-gz-sim
```

### What Have You Tried?

```shell

```

### Relevant Logs / Terminal Output

```shell

```

GyuH13, 2025-04-09:
Hello @brean
I understand the issue.
I will work on creating a separate `TurtleBot3 Navigation2 launch`.
Thank you for the clarification.

GyuH13, 2025-05-16:
@brean 
Thank you for waiting. After internal review, we’ve concluded that creating our own navigation launch files would lead to differences between the Humble and Jazzy versions. Also, it would be difficult for us to reflect updates from the official navigation repository, so creating our own launch files may not be feasible.

I personally recommend that you create a navigation launch file within the TurtleBot package yourself, so that you can use it without installing the nav2_bringup package. If you encounter any difficulties in doing so, I will help you!
If you have any question about these, please reply in here. Thanks.

brean, 2025-05-16:
Hey @GyuH13 ,
the problem is that the dependency forces you to install all graphics dependencies (including gazebo) on the robot itself when you install the packages. Note that I didn't and don't want to touch any turtlebot3-code.
I've only build the packages and if you want to provide the packages for the turtlebot at any point in the future you'll encounter the same problem.
Interestingly you ignore the problem by just not using the debian-packages, just cloning the turtlebot3-repo and deleting the navigation-folder: https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/
So just not installing the debian packages on the robot would also work.
However I would love to see debian packages that you can install from the official ros-humble instead of building it on the robot which saves a lot of time, especially with only the 2 GB memory.