ROBOTIS-GIT/turtlebot3 #1096: Wifi connection issue while setting up SBC

Harshul-18, 2025-05-04:
### Operating System:

Ubuntu 20.04

### ROS version :

Noetic

### Turtlebot3 Model:

Burger

### Description

I'm having issues connecting my Raspberry Pi 3 running Ubuntu (installed as instructed in the [Documentation](https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup)) to WiFi. 

### Steps to reproduce issue

1. Flashed [Turtlebot3_rp3_image](https://www.robotis.com/service/download.php?no=2008&_gl=1*mueswd*_gcl_au*NDIxNzY1NTIwLjE3NDQ1NTU1NDc.) using Raspberry Pi Imager
2. First tried using default settings, and then configured WiFi credentials during imaging process
3. Booted Raspberry Pi and logged in
4. Checked WiFi status with `ip addr show wlan0` - shows NO-CARRIER, and command `hostname -I` shows no connectivity to a wifi.
5. Edited netplan configuration at /etc/netplan/50-cloud-init.yaml as instructed in one of the [previous issue](https://github.com/ROBOTIS-GIT/turtlebot3/issues/700#issuecomment-780373114)
6. Also tried manual connection with `wpa_supplicant.conf` as in [issue](https://github.com/ROBOTIS-GIT/turtlebot3/issues/700#issuecomment-779384287)

### What Have You Tried?

```shell
- Edited netplan configuration at /etc/netplan/50-cloud-init.yaml as instructed in one of the [previous issue](https://github.com/ROBOTIS-GIT/turtlebot3/issues/700#issuecomment-780373114)
- Also tried manual connection with `wpa_supplicant.conf` as in [issue](https://github.com/ROBOTIS-GIT/turtlebot3/issues/700#issuecomment-779384287)
- And after each change, i tried rebooting the system.
```

### Relevant logs / Terminal output

```shell

```

### Screenshots or Diagrams

`/etc/netplan/50-cloud-init.yaml` file
[image]

`/etc/wpa_supplicant/wpa_supplicant.conf` file

[image]

GyuH13, 2025-05-07:
Hello @Harshul-18 
If you can't connect to Wi-Fi through e - manual, consider using the Raspberry Pi Imager to directly install the 20.04 image. You can configure the Wi-Fi settings before installation through the setup options. For more details, please refer to the link below and proceed with the 20.04 version. 
[robotis sbc setting emanual](https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#install-ubuntu-2204)

GyuH13, 2025-05-20:
Hello @Harshul-18 
Sorry for the late reply.

There isn't another separate custom image provided, but how about installing the official Ubuntu 20.04 image using the Raspberry Pi Imager? You can configure Wi-Fi settings through the Imager's setup options. This method is the most reliable when it comes to Wi-Fi connection issues. I’ve also had cases where Wi-Fi wouldn’t connect when setting it up manually. Sometimes it only connected after several reboots. That’s why I recommend installing the official image and configuring Wi-Fi during the setup.
Once you've installed the official image, you can follow the Humble version tutorial step by step to install the necessary packages. But make sure to install the Noetic version instead. (For example, install ROS Noetic, TurtleBot packages compatible with the Noetic version, etc.)

Harshul-18, 2025-05-25:
Greetings of the day @GyuH13 ,

I have tried installing Ubuntu 20.04 from rpi imager, but that gave me "Could Not Resolve Host" error, and I tried to search ubuntu 20.04 images online, but none of them got installed, rpi imager showed error "exFAT file system not supported". Is there any ros-noetic compatible ubuntu 20.04 image official source available currently?

[image]

GyuH13, 2025-09-10:
Sorry for the late reply. @Harshul-18 
You can solve that problem to format the SD card with the FAT32 file system. 
First, use `$ lsblk` to identify the device name (e.g., /dev/sdb),
then run: `$ sudo mkfs.vfat -F 32 /dev/sdb` to format the SD card to FAT32. 
After that, you can verify the FAT32 file system by running `$ lsblk -f`

Also, regarding the “Could Not Resolve Host” error, does this occur when downloading the image file we provide?