ROBOTIS-GIT/turtlebot3 #1097: LaserScan drifts on map frame as robot moves, causing navigation to fail

hyeonji0401, 2025-05-07:
### Operating System:

Ubuntu22.04

### ROS version :

humble

### Turtlebot3 Model:

burger, lds-02 , jetson orin nano

### Description

I'm using ROS 2 Humble with `turtlebot3_bringup` and `turtlebot3_navigation2`. The robot starts up correctly, and AMCL localization is active with a static map.

The initial pose is set correctly using the "2D Pose Estimate" tool in RViz, and the robot appears properly localized at startup.

However, the issue is that as the robot moves, the LaserScan data (`/scan`) appears to drift relative to the static map in RViz — even though the environment is completely static. The scan points begin to misalign from the walls as the robot moves through the map.

As a result, navigation fails. The robot is unable to reach goals, often generates invalid paths, or gets stuck. The costmap and the map are no longer in sync.

Clarifications:
- The LaserScan data itself is fine in the robot's local frame.
- RViz’s Fixed Frame is set to `map`.
- TF tree (`base_link → odom → map`) appears structurally correct.
- The issue seems to be in how transforms are applied or possibly AMCL drift.

Additionally, the robot tends to move in the wrong direction (opposite to the path), and it struggles especially when the goal is behind the robot. I attempted editing the URDF for orientation-related fixes, but it didn't help.

Is this a known TF synchronization issue or AMCL misconfiguration?
Any advice on how to debug or resolve this properly would be appreciated.

### Steps to reproduce issue

1. Launch `turtlebot3_bringup` on the TurtleBot3 (Jetson orin nano).
2. On a remote PC, launch `turtlebot3_navigation2` with a saved static map and AMCL.
3. In RViz:
   - Set the Fixed Frame to `map`
   - Use "2D Pose Estimate" to set the robot’s initial pose on the map.
   - Use "2D Nav Goal" to send a navigation goal.
4. Observe `/scan` data and map alignment in RViz.
5. As the robot begins to move:
   - The scan data appears to drift or shift relative to the static map.
   - The robot cannot follow the planned path accurately.
   - Navigation either fails entirely, or the robot gets stuck.

### What Have You Tried?

```shell
- I have modified `turtlebot3_navigation2/param/burger.yaml` with various parameter tuning attempts (AMCL, costmap settings, inflation radius, etc.).
- I tried switching the middleware from Fast DDS to Cyclone DDS to check for TF or timing-related issues.
- I verified the TF tree using `ros2 run tf2_tools view_frames` and confirmed that `map → odom → base_link` exists and appears valid.
- I also confirmed that `/scan` data looks fine in the robot's local frame
- I also attempted modifying the URDF to address a possible orientation issue, as the robot tends to move in the wrong direction or opposite to the path.
- The robot especially struggles to reach goals that are positioned behind it, which suggests a possible issue with orientation control or local planner behavior.
```

### Relevant logs / Terminal output

```shell
[component_container_isolated-1] [INFO] [1746537016.648914360] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537017.648943753] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537018.648946838] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537019.749014645] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537020.748939587] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537021.748991170] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537022.748937273] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537023.848937454] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537024.848940115] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [ERROR] [1746537025.549056413] [controller_server]: Failed to make progress
[component_container_isolated-1] [WARN] [1746537025.549242400] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[component_container_isolated-1] [INFO] [1746537025.568467374] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[component_container_isolated-1] [INFO] [1746537025.571349977] [controller_server]: Received a goal, begin computing control effort.
[component_container_isolated-1] [WARN] [1746537025.721816211] [controller_server]: Control loop missed its desired rate of 10.0000Hz
[component_container_isolated-1] [INFO] [1746537025.871874891] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537026.871988920] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537027.971872681] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537028.971867104] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537029.972202416] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537030.971822974] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537032.071815501] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537033.071789760] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537034.071809317] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [INFO] [1746537035.171833617] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [ERROR] [1746537035.771846909] [controller_server]: Failed to make progress
[component_container_isolated-1] [WARN] [1746537035.771887080] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[component_container_isolated-1] [WARN] [1746537035.798232395] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.
[component_container_isolated-1] [ERROR] [1746537035.798342215] [bt_navigator]: Goal failed
```

### Screenshots or Diagrams

[image]

[image]

GyuH13, 2025-05-09:
Hello @hyeonji0401 
We're sorry to hear that you've experienced inconvenience while using TurtleBot3.

First of all, it's a bit difficult to fully understand the situation based on the text alone. If possible, could you please share a screen recording or video of the issue? This would greatly help us in identifying and resolving the problem.

Thanks

hyeonji0401, 2025-05-09:
@GyuH13  

Hi, I'm not a native English speaker and this is my first time using TurtleBot, so I apologize if my explanation is unclear. 😂
Here's a video link  [video](https://www.youtube.com/watch?v=_A9Rst7-IOI)    I would appreciate it if you could take a look. Thank you! 😊

GyuH13, 2025-05-16:
@hyeonji0401 
I'm glad to hear the issue has been resolved. I'm sorry I couldn't be of more help this time.
If any other issues come up, please feel free to reach out — I'll do my best to assist you. Thank you!
And I will close this issue.

closed 2025-05-16 by GyuH13