ROBOTIS-GIT/turtlebot3 #1098: Error) ros2 launch turtlebot3_gazebo turtlebot3_house.launch.py

kimudonghyeon, 2025-05-16:
### Operating System:

ubuntu 22.04

### ROS version :

humble

### Turtlebot3 Model:

burger

### Description

When I run empty_world.launch.py or turtlebot3_world.launch.py, Gazebo runs normally.
But when I run turtlebot3_house.launch.py, Gazebo does not start and keeps loading.
I installed ROS2 and the TurtleBot3 package following the e-manual on May 16th.

### Steps to reproduce issue

It is the same as what I wrote above.

### What Have You Tried?

```shell
At first, I couldn't run 'ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py', too.
So i entered
1. ros2 pkg list | grep turtlebot3_gazebo
2. sudo apt-get update
3. sudo apt-get install ros-humble-turtlebot3-gazebo

Finally i could run 'ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py'.
However, i can't run 'ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py'still.
```

### Relevant logs / Terminal output

```shell
[INFO] [launch]: All log files can be found below /home/kim/.ros/log/2025-05-16-14-11-35-998441-kim-MS-7D42-73655
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
urdf_file_name : turtlebot3_burger.urdf
[INFO] [gzserver-1]: process started with pid [73656]
[INFO] [gzclient-2]: process started with pid [73658]
[INFO] [robot_state_publisher-3]: process started with pid [73660]
[INFO] [spawn_entity.py-4]: process started with pid [73662]
[robot_state_publisher-3] [INFO] [1747372296.535619245] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1747372296.535666898] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1747372296.535670758] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-3] [INFO] [1747372296.535672991] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-3] [INFO] [1747372296.535675136] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1747372296.535677272] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-3] [INFO] [1747372296.535679320] [robot_state_publisher]: got segment wheel_right_link
[spawn_entity.py-4] [INFO] [1747372296.721266553] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1747372296.721426410] [spawn_entity]: Loading entity XML from file /opt/ros/humble/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf
[spawn_entity.py-4] [INFO] [1747372296.721827488] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1747372296.721945453] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1747372297.725038600] [spawn_entity]: Calling service /spawn_entity
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-3] [INFO] [1747372303.307657129] [rclcpp]: signal_handler(signum=2)
[gzserver-1] [INFO] [1747372303.307966065] [rclcpp]: signal_handler(signum=2)
[spawn_entity.py-4] Traceback (most recent call last):
[spawn_entity.py-4]   File "/opt/ros/humble/lib/gazebo_ros/spawn_entity.py", line 372, in <module>
[spawn_entity.py-4]     main()
[spawn_entity.py-4]   File "/opt/ros/humble/lib/gazebo_ros/spawn_entity.py", line 367, in main
[spawn_entity.py-4]     exit_code = spawn_entity_node.run()
[spawn_entity.py-4]   File "/opt/ros/humble/lib/gazebo_ros/spawn_entity.py", line 228, in run
[spawn_entity.py-4]     success = self._spawn_entity(entity_xml, initial_pose, spawn_service_timeout)
[spawn_entity.py-4]   File "/opt/ros/humble/lib/gazebo_ros/spawn_entity.py", line 290, in _spawn_entity
[spawn_entity.py-4]     rclpy.spin_once(self)
[spawn_entity.py-4]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 206, in spin_once
[spawn_entity.py-4]     executor.spin_once(timeout_sec=timeout_sec)
[spawn_entity.py-4]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 751, in spin_once
[spawn_entity.py-4]     self._spin_once_impl(timeout_sec)
[spawn_entity.py-4]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 740, in _spin_once_impl
[spawn_entity.py-4]     handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
[spawn_entity.py-4]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 723, in wait_for_ready_callbacks
[spawn_entity.py-4]     return next(self._cb_iter)
[spawn_entity.py-4]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 620, in _wait_for_ready_callbacks
[spawn_entity.py-4]     wait_set.wait(timeout_nsec)
[spawn_entity.py-4] KeyboardInterrupt
[ERROR] [spawn_entity.py-4]: process has died [pid 73662, exit code -2, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity burger -file /opt/ros/humble/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf -x -2.0 -y -0.5 -z 0.01 --ros-args'].
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 73660]
[ERROR] [gzclient-2]: process[gzclient-2] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[ERROR] [gzserver-1]: process[gzserver-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [gzclient-2]: sending signal 'SIGTERM' to process[gzclient-2]
[INFO] [gzserver-1]: sending signal 'SIGTERM' to process[gzserver-1]
[gzserver-1] [INFO] [1747372308.323981823] [rclcpp]: signal_handler(signum=15)
[ERROR] [gzclient-2]: process has died [pid 73658, exit code -15, cmd 'gzclient --gui-client-plugin=libgazebo_ros_eol_gui.so'].
[ERROR] [gzserver-1]: process[gzserver-1] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [gzserver-1]: sending signal 'SIGKILL' to process[gzserver-1]
[ERROR] [gzserver-1]: process has died [pid 73656, exit code -9, cmd 'gzserver /opt/ros/humble/share/turtlebot3_gazebo/worlds/turtlebot3_house.world -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].
```

### Screenshots or Diagrams

[image]

kimudonghyeon, 2025-05-16:
Also, colcon build:--- stderr: turtlebot3_gazebo                                        
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:72 (include)
  CMakeLists.txt:23 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---

GyuH13, 2025-05-16:
@kimudonghyeon Hello.
The first time you launch the TurtleBot3 House simulation, it may take a long time (more than 5 minutes) to download and install necessary resources. Would you mind waiting about 10 minutes?

GyuH13, 2025-06-08:
@kimudonghyeon Hello
Have you resolved the issue?
Since there has been no further discussion, I will consider the issue resolved and proceed to close it.
If you have any further issues or questions, please feel free to reopen the issue.

closed 2025-06-08 by GyuH13